karstx/sinowisp #8: Doesn't work on Windows (tested with Air96)

WinkelCode, 2023-08-13:
```
2023-08-13T21:16:57.963Z INFO  [sinowealth_kb_tool::isp] Found Device. Entering ISP mode...
2023-08-13T21:16:57.965Z INFO  [sinowealth_kb_tool::isp] Waiting for bootloader device...
2023-08-13T21:16:58.993Z INFO  [sinowealth_kb_tool::isp] Device didn't come up...
2023-08-13T21:16:58.993Z INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
2023-08-13T21:16:59.016Z INFO  [sinowealth_kb_tool::isp] Found Device. Entering ISP mode...
2023-08-13T21:16:59.016Z INFO  [sinowealth_kb_tool::isp] Waiting for bootloader device...
2023-08-13T21:17:00.035Z INFO  [sinowealth_kb_tool::isp] Device didn't come up...
2023-08-13T21:17:00.035Z INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 3/10
[and so on...]
```

When I use my tool to set it to ISP mode:
```
2023-08-13T21:17:33.342Z INFO  [sinowealth_kb_tool::isp] No KB found. Trying bootloader directly...
2023-08-13T21:17:33.389Z INFO  [sinowealth_kb_tool::isp] Connected!
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: InvalidParam', src/isp.rs:169:55
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

This is a Python version of the system from Nudelta to find the "channels" to communicate with the keyboard, in this case configured to set the device into ISP mode: https://gist.github.com/WinkelCode/e4e16dfed3ce6f146b5dcfad2943ff72

carlossless, 2023-08-17:
@WinkelCode thanks for the report ✌️, I'll check this and see what needs to be fixed a bit later.

carlossless, 2023-08-27:
This should be resolved now that https://github.com/carlossless/sinowealth-kb-tool/pull/10 is merged and [v0.0.3](https://github.com/carlossless/sinowealth-kb-tool/releases/tag/v0.0.3) released.

closed 2023-08-27 by carlossless