karstx/sinowisp #16: Can't enter ISP mode

cloudenum, 2023-11-14:
Hello,

I've forked this project to add my own keyboard which is a Digitial Alliance Meca Warrior X. It uses SH68F90S MCU as indicated in the chip marking below.

[image]

What I did is added this PART configuration 
```rust
pub const PART_DA_WARRIOR_X: Part = Part {
    flash_size: 61440, // 61440 until bootloader
    bootloader_size: 4096,
    page_size: 2048,
    vendor_id: 0x258a,
    product_id: 0x0090,
};
```

I got the Vendor ID and Product ID from Windows Device Manager, but I don't know about the flash_size, bootloader_size and page_size.

Anyway here's the log:
```log
2023-11-14T16:52:41.155Z INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0090
2023-11-14T16:52:41.155Z DEBUG [sinowealth_kb_tool::isp] Opening: "\\\\?\\HID#VID_258A&PID_0090&MI_01&Col05#8&2c5f1391&0&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}"
2023-11-14T16:52:41.168Z INFO  [sinowealth_kb_tool::isp] Found Regular device. Entering ISP mode...
2023-11-14T16:52:41.169Z INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
2023-11-14T16:52:41.861Z INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
```

I have little experience with embedded programming (mostly arduino).

If you able to assist me, I would love to help and get this tool working on this keyboard and hopefully able to flash QMK Firmware.

carlossless, 2023-11-18:
Hey @cloudenum,

Thanks for the device report! Your part configuration looks okay - it's what I would expect for sh68f90 parts. Though, since sinowealth-kb-tool is not working for you, I am guessing that there's either no ISP bootloader on your device or no way (or an unknown way) to launch into it...

For now, I'd ask you the same as I did in https://github.com/carlossless/sinowealth-kb-tool/issues/15#issuecomment-1799563270

> hopefully able to flash QMK Firmware.

It's very unlikely QMK will be ported on 8051-based devices like this one. Popular and full-featured compiler suites like gcc and llvm don't target this architecture and sdcc lacks a number of features to build the current QMK sources.

I am working on an alternative QMK-like firmware [here](https://github.com/carlossless/sh68f90a-experiments) though, so you might want to stay tuned for that.

cloudenum, 2023-11-21:
Okay, I will try to dump the firmware when I picked up my Arduino.

gashtaan, 2023-12-16:
@carlossless I guess your tool doesn't work in this case because you have hard-coded the tool to work with HID device with path which contains Col05. This way it doesn't work for any of my keyboards too. The ColXX simply doesn't mean anything, Windows OS just assign some number to collections in interface descriptors. I think that more generic approach would be to choose the device where usage = 1 and usage_page >= 0xFF00 (vendor specific).

carlossless, 2023-12-20:
@gashtaan thanks for the suggestion, I have made the proposed changes in #26 and published a new version, which should work.

@cloudenum would you mind giving it another try?

cloudenum, 2024-01-06:
Thanks for the update @carlossless now reading both firmware and bootloader works on my keyboard, but I haven't tried writing to it.

Here's the firmware and bootloader dumped by this tool, if you're interested.
[dump files](https://gist.github.com/cloudenum/ea52843a3ae6dbf4963f0853deb410eb)

carlossless, 2024-01-06:
Thanks for the report and for sharing the dumps!

closed 2024-01-06 by carlossless