karstx/sinowisp #25: [device report] Royal Kludge RK100

namidairo, 2023-12-20:
```rust
pub const PART_ROYALKLUDGE_RK100: Part = Part {
    firmware_size: 61440, // 61440 until bootloader
    bootloader_size: 4096,
    page_size: 2048,
    vendor_id: 0x258a,
    product_id: 0x0056,
};
```
```
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0056
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0056
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Connected!
INFO  [sinowealth_kb_tool::isp] Enabling firmware...
INFO  [sinowealth_kb_tool::isp] Reading...
INFO  [sinowealth_kb_tool] MD5: cfc8661da8c9d7e351b36c0a763426aa
```
Product URL: `http://en.rkgaming.com/product/14/`
This is the RGB variant of the product, but I suspect that the backlit and the Pro variants of the product are using the same mcu. (I'd have to check their firmware updaters to be sure)

I note that the bootloader md5 is the same as the Xinmeng K916, but I don't remember the precise label on the mcu at this time.

Side note: Couldn't put the keyboard into ISP mode while within Windows, the `05 75 ...` report would never get sent. USB captures in Wireshark seem to corroborate this. (The vendor's firmware updater binary wouldn't detect either)

Possibly similar issue to #15

carlossless, 2023-12-20:
@namidairo thanks for the report, I just published fixes that should address the issue you experienced on Windows. Would you mind giving it another try?

carlossless, 2023-12-20:
@namidairo Also am I right in assuming you tested reading only?

namidairo, 2023-12-21:
Tested reading only.

New commits appear to have done the job, can read out the bootloader in Windows now. (isp_index 0 worked here)

carlossless, 2023-12-21:
Seems like at least some variants of the RK100 have sh68f90* ICs with the full sinowealth labeling (not BYK***) - https://youtu.be/hrh6N3uzNhc?si=C9_hAEDYw2q3et19&t=733. Would be great to know which chip is it exactly though!

carlossless, 2023-12-21:
@namidairo

> This particular one appears to be labelled BYK916

thanks!

namidairo, 2023-12-23:
>  Do you happen to know if these older models used a different / differently labeled MCU, or was it also a SH68F90* / BYK916?

Pretty sure they were older board revisions of various models that used Sonix (or HFD rebranded) MCU, was just noting their convenient to access location. Apologies if I got your hopes up.

carlossless, 2023-12-23:
> Pretty sure they were older board revisions of various models that used Sonix (or HFD rebranded) MCU, was just noting their convenient to access location. Apologies if I got your hopes up.

Ah well, no worries! I was curious because there might be a possibility to trigger the ISP bootloader via the USB port on SH68F90 (not a USB packet). @gashtaan has found it and looking into using it.

closed 2023-12-23 by carlossless