karstx/sinowisp #28: [device-report] Royal Kuludge RK71

gauravmanmode, 2023-12-21:
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00ea
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00ea
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 3/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00ea
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 4/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00ea
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Connected!
INFO  [sinowealth_kb_tool::isp] Enabling firmware...
INFO  [sinowealth_kb_tool::isp] Reading...
INFO  [sinowealth_kb_tool] MD5: e5277ac8658db28e33b3542aa06e4136

not tested writing

gauravmanmode, 2023-12-21:
My keyboard is bricked.
I was just messing around and i tried the write option, now it connects with 
HID\VID_05AC&PID_024F&REV_0000&MI_01&Col04 
some green and blue lights are constantly on
any advice on what i should do now?

carlossless, 2023-12-21:
@gauravmanmode if you can still write to your keyboard, that means it's still capable of booting into ISP mode.

Could you please make a full dump of your keyboard with the `--full` option and send it to me contact[at]carlossless.io. I'll try inspecting it and see what's wrong with it.

gauravmanmode, 2023-12-21:
No, the keyboard connects as a Keyboard HID device with above parameters, keys are missassigned and pressing any key just sends random values. The lights are constant blue and green combination. 
When i try the sinowealth-kb-tool, it says regular device found ,Trying ISP device but thats all.

carlossless, 2023-12-21:
@gauravmanmode thanks for sending me the dump, although it only includes the main firmware part and not the bootloader.

Regardless, and just to confirm, your keyboard got bricked after executing only the **read** operation? Or was it after you **wrote** something to it?

If it's the latter what did you write to it?

gauravmanmode, 2023-12-23:
Actually, I had compiled smk locally with some changes in layout and other minor changes and i wrote that. The keyboard connects as a HID device. Is the bootloader still intact or have i messed up? Can i not use the JTAG pins to flash the original one.

carlossless, 2023-12-23:
Oh ok :)

I'm guessing that the way your keyboard is connected together is different from how nuphy-air60 is (which is what SMK currently supports). I assume that's why you might be getting different key presses/random rgb lighting, but I'm not certain.

If the keyboard connects as an HID device it should still be possible to boot into ISP mode and flash the original firmware that you dumped. Have you already tried that? For SMK it should be `isp_index = 0`

> Can i not use the JTAG pins to flash the original one.

You can, but you'll need a sinolink device and the ProWriter tool to do that. You'll also need to make sure you reconstruct the full firmware image (main firmware + bootloader) and adjust/remove the [changes the ISP bootloader does](https://github.com/carlossless/sinowealth-kb-tool?tab=readme-ov-file#reading) before you flash it over.

gauravmanmode, 2024-01-07:
Hi, finally i was able to fix my keyboard.
Actually, the device doesn't enter ISP mode in Windows shows
Regular device found ,Trying ISP device but worked fine with ubuntu 22
Here's the log

ubuntu@ubuntu:~/Downloads/sinowealth-kb-tool-x86_64-unknown-linux-gnu-v0.0.7$ sudo ./sinowealth-kb-tool read -p nuphy-air60 firmware.hex
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x05ac pId:0x024f
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Connected!
INFO  [sinowealth_kb_tool::isp] Enabling firmware...
INFO  [sinowealth_kb_tool::isp] Reading...
INFO  [sinowealth_kb_tool] MD5: 852c8a14415d783ff7d6a206faa41804
ubuntu@ubuntu:~/Downloads/sinowealth-kb-tool-x86_64-unknown-linux-gnu-v0.0.7$ sudo ./sinowealth-kb-tool read -p nuphy-air60 -b bootloader.hex
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x05ac pId:0x024f
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Connected!
INFO  [sinowealth_kb_tool::isp] Enabling firmware...
INFO  [sinowealth_kb_tool::isp] Reading...
INFO  [sinowealth_kb_tool] MD5: cfc8661da8c9d7e351b36c0a763426aa
ubuntu@ubuntu:~/Downloads/sinowealth-kb-tool-x86_64-unknown-linux-gnu-v0.0.7$ sudo ./sinowealth-kb-tool read -p nuphy-air60 --full full.hex
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x05ac pId:0x024f
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Connected!
INFO  [sinowealth_kb_tool::isp] Enabling firmware...
INFO  [sinowealth_kb_tool::isp] Reading...
INFO  [sinowealth_kb_tool] MD5: 845f0726c859057584d3ecffc429973e
ubuntu@ubuntu:~/Downloads/sinowealth-kb-tool-x86_64-unknown-linux-gnu-v0.0.7$ sinowealth-kb-tool write -p nuphy-air60 a.hex
sinowealth-kb-tool: command not found
ubuntu@ubuntu:~/Downloads/sinowealth-kb-tool-x86_64-unknown-linux-gnu-v0.0.7$ sudo ./sinowealth-kb-tool write -p nuphy-air60 a.hex
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x05ac pId:0x024f
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Connected!
INFO  [sinowealth_kb_tool::isp] Erasing...
INFO  [sinowealth_kb_tool::isp] Writing...
INFO  [sinowealth_kb_tool::isp] Reading...
INFO  [sinowealth_kb_tool::isp] Verifying...
INFO  [sinowealth_kb_tool::isp] Enabling firmware...

I successfully flashed the old firmware i had and now its working.
Thanks for the hard work.

gauravmanmode, 2024-01-07:
Also, on ubuntu 22, version 0.0.6 and 0.0.7 do not seem to work, only Trying to find ISP device,
whereas works flawlessly with 0.0.5 both read and write.
also flashing old firmware seems to have changes hash

ghm@ghm-HP-Pavilion-g6-Notebook-PC:/Documents/sinowealth-kb-tool-x86_64-unknown-linux-gnu-v0.0.5$ sudo ./sinowealth-kb-tool read  --vendor_id 0x258a  --product_id 0x00ea  --firmware_size 61440 --bootloader_size 4096 --page_size 2048 --isp isp.hex 
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00ea
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00ea
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Connected!
INFO  [sinowealth_kb_tool::isp] Enabling firmware...
INFO  [sinowealth_kb_tool::isp] Reading...
INFO  [sinowealth_kb_tool] MD5: cfc8661da8c9d7e351b36c0a763426aa
ghm@ghm-HP-Pavilion-g6-Notebook-PC:/Documents/sinowealth-kb-tool-x86_64-unknown-linux-gnu-v0.0.5$ sudo ./sinowealth-kb-tool read  --vendor_id 0x258a  --product_id 0x00ea  --firmware_size 61440 --bootloader_size 4096 --page_size 2048 --full full.hex 
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00ea
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Connected!
INFO  [sinowealth_kb_tool::isp] Enabling firmware...
INFO  [sinowealth_kb_tool::isp] Reading...
INFO  [sinowealth_kb_tool] MD5: 9a2726ada41114972d63531fd823514c
ghm@ghm-HP-Pavilion-g6-Notebook-PC:/Documents/sinowealth-kb-tool-x86_64-unknown-linux-gnu-v0.0.5$ sudo ./sinowealth-kb-tool read  --vendor_id 0x258a  --product_id 0x00ea  --firmware_size 61440 --bootloader_size 4096 --page_size 2048 firmware.hex 
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00ea
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Connected!
INFO  [sinowealth_kb_tool::isp] Enabling firmware...
INFO  [sinowealth_kb_tool::isp] Reading...
INFO  [sinowealth_kb_tool] MD5: 2c3da23346f939deb2d6a1f4945f0bc6

carlossless, 2024-01-07:
@gauravmanmode Hey, thanks for sharing the bootloader md5sum and I'm glad to see you got it to work, although I am a bit confused about the issues you reported.

[Here](https://github.com/carlossless/sinowealth-kb-tool/issues/28#issuecomment-1879990847) you mentioned that you got it to work on ubuntu 22 and the logs you attached reference version v0.0.7, but [here](https://github.com/carlossless/sinowealth-kb-tool/issues/28#issuecomment-1880002380) you mentioned that v0.0.7 and v0.0.6 didn't work on ubuntu 22 and only version v0.0.5 did. Perhaps there was a mistake somewhere?

I suspect that it didn't work on windows because it needs a different `--isp_index` than the default provided to it. Would you mind sharing what you see by running [`usbhid-dump`](https://github.com/DIGImend/usbhid-dump) for your keyboard device?

gauravmanmode, 2024-01-09:
Hi, actually v 0.0.7 and v 0.0.6 do not work and v0.0.5 works for me when the firmware on the keyboard is the original RK 
whereas when smk is on the keyboard v0.0.7 works.
Here is the attached log of usbhid-dump

001:003:000:DESCRIPTOR         1704739610.940652
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 95 05 75 01 81 02 95 01 75 03 81 01
 05 01 09 30 09 31 16 00 F8 26 FF 07 75 0C 95 02
 81 06 09 38 15 81 25 7F 75 08 95 01 81 06 C0 C0

003:003:001:DESCRIPTOR         1704739610.942788
 05 01 09 80 A1 01 85 01 19 81 29 83 15 00 25 01
 75 01 95 03 81 02 95 05 81 01 C0 05 0C 09 01 A1
 01 85 02 19 00 2A 3C 02 15 00 26 3C 02 95 01 75
 10 81 00 C0 06 00 FF 09 01 A1 01 85 05 15 00 26
 FF 00 19 01 29 02 75 08 95 05 B1 02 C0 05 01 09
 06 A1 01 85 06 05 07 19 04 29 70 15 00 25 01 75
 01 95 78 81 02 C0 06 00 FF 09 01 A1 01 85 0A 15
 00 26 FF 00 09 00 75 08 95 40 B1 02 C0

003:003:000:DESCRIPTOR         1704739610.944209
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 95 08 75 01 81 02 95 01 75 08 81 03 95 06 75 08
 15 00 26 FF 00 05 07 19 00 2A FF 00 81 00 25 01
 95 05 75 01 05 08 19 01 29 05 91 02 95 01 75 03
 91 03 C0

carlossless, 2024-01-13:
@gauravmanmode hey, thanks for the usbhid-dump and the firmware dump you sent me earlier. I figured out what the issue with the tool was (details in #43) and released a new version - https://github.com/carlossless/sinowealth-kb-tool/releases/tag/v0.0.8. Please test it out, it should work both on linux and windows!

gauravmanmode, 2024-01-14:
Thanks, it is working both linux and windows.

closed 2024-01-14 by gauravmanmode