karstx/sinowisp #29: [device-report] Weikav Sugar65

normano64, 2023-12-21:
The Weikav Sugar65 is using a SinoWealth SH68F90S and use the same vendor_id and product_id as nuphy-air60 and I got and it
```
.\sinowealth-kb-tool-0.0.5.exe read -p nuphy-air60
         MD5: ba3592aa17b5549f2bf72a06d902a764
-b       MD5: 2d169670eae0d36eae8188562c1f66e8
--full   MD5: e19cdb907ac25f825a7caf8f0f88a9f9
```
[image]

The 0.0.6 version didn't work for me on windows so I tried the 0.0.5 version which worked
0.0.6:
```
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x05ac pId:0x024f
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
```
0.0.5:
```
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x05ac pId:0x024f
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Waiting for ISP device...
INFO  [sinowealth_kb_tool::isp] Connected!
```

normano64, 2023-12-22:
@carlossless Thanks for the answer. I tried the above in both windows without success. I tried wsl2 with usbipd-win for usb pass-through (over ip) but it gave the same results as on windows

I got the following lsusb output for the device in wsl2 if it is of any help

```
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x05ac Apple, Inc.
  idProduct          0x024f Aluminium Keyboard (ANSI)
  bcdDevice            0.26
  iManufacturer           1 SINO WEALTH
  iProduct                2 Mechanical Keyboard
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x003b
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      89
          Report Descriptor: (length is 89)
            Item(Global): Usage Page, data= [ 0x01 ] 1
                            Generic Desktop Controls
            Item(Local ): Usage, data= [ 0x06 ] 6
                            Keyboard
            Item(Main  ): Collection, data= [ 0x01 ] 1
                            Application
            Item(Global): Usage Page, data= [ 0x07 ] 7
                            Keyboard
            Item(Local ): Usage Minimum, data= [ 0xe0 ] 224
                            Control Left
            Item(Local ): Usage Maximum, data= [ 0xe7 ] 231
                            GUI Right
            Item(Global): Logical Minimum, data= [ 0x00 ] 0
            Item(Global): Logical Maximum, data= [ 0x01 ] 1
            Item(Global): Report Count, data= [ 0x08 ] 8
            Item(Global): Report Size, data= [ 0x01 ] 1
            Item(Main  ): Input, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Global): Report Count, data= [ 0x01 ] 1
            Item(Global): Report Size, data= [ 0x08 ] 8
[... 71 lines cut ...]
      bInterfaceProtocol      0
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     229
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

carlossless, 2023-12-23:
@normano64 thanks for sharing that output, it's very helpful that it contains the HID report descriptor, but it makes it seem that Col05 would be the collection with `usage_page == 0xff00 && usage == 0x0003`, though it only accepts one byte?

Anyway, this tells me that I can't rely on the `usage_page` and `usage` being the same for all of these different keyboards and will need to expose that as customizable option.

Would you mind running [win-hid-dump](https://github.com/todbot/win-hid-dump) and sharing the output (all records that concern your device)?

normano64, 2023-12-23:
@carlossless The win-hid-dump throws an exception so I don't know if there are more entries for the keyboard
```
05AC:024F: SINO WEALTH - Mechanical Keyboard
PATH:\\?\hid#vid_05ac&pid_024f&mi_00#8&2cfe8627&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
DESCRIPTOR:
  05  01  09  06  A1  01  05  07  19  E0  29  E7  15  00  25  01
  35  00  45  01  65  00  55  00  75  01  95  08  81  02  95  30
  81  03  05  FF  09  03  25  FF  45  00  75  08  95  01  81  02
  05  08  19  01  29  05  25  01  45  01  75  01  95  05  91  02
  95  03  91  03  05  0C  09  00  25  7F  45  00  75  08  95  40
  B1  02  C1  00
  (84 bytes)
```

A github comment suggested wsl2 usbhid-dump which gave me 
```
001:003:001:DESCRIPTOR
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 05 07 19 00 29 67 15 00
 25 01 75 01 95 68 81 02 09 85 09 87 09 88 09 89
 09 8A 09 8B 09 91 09 92 75 01 95 08 81 02 C0 05
 0C 09 01 A1 01 85 02 19 00 2A 3C 02 15 00 26 3C
 02 95 01 75 10 81 00 C0 05 01 09 02 A1 01 85 03
 09 01 A1 00 05 09 19 01 29 05 15 00 25 01 75 01
 95 05 81 02 95 03 81 01 05 01 09 30 09 31 16 00
 80 26 FF 7F 75 10 95 02 81 06 09 38 15 80 25 7F
 75 08 95 01 81 06 C0 C0 06 00 FF 09 01 A1 01 85
 04 15 00 26 FF 00 09 30 75 08 95 08 81 02 09 31
 91 02 C0 06 00 FF 09 02 A1 01 85 05 15 00 26 FF
 00 19 01 29 02 75 08 95 05 B1 02 C0 06 01 FF 09
 01 A1 01 85 07 15 00 25 FF 19 01 29 02 75 08 96
 07 02 B1 02 C0

001:003:000:DESCRIPTOR
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 95 08 75 01 81 02 95 01 75 08 81 01 95 05 75 01
 05 08 19 01 29 05 91 02 95 01 75 03 91 01 05 07
 19 00 2A FF 00 95 05 75 08 15 00 26 FF 00 81 00
 05 FF 09 03 75 08 95 01 81 02 05 0C 09 00 15 80
 25 7F 95 40 75 08 B1 02 C0
 ```

carlossless, 2023-12-23:
Ah great thank you!

Yeah, I'm not sure why lsusb only had a part of the descriptor, but from the usbhid-dump I can tell that the ISP report lives under `usage_page == 0xff00 && usage == 0x0002` instead of `usage == 0x0001`:

```
0x06, 0x00, 0xFF,  // Usage Page (Vendor Defined 0xFF00)
0x09, 0x02,        // Usage (0x02)
0xA1, 0x01,        // Collection (Application)
0x85, 0x05,        //   Report ID (5)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x19, 0x01,        //   Usage Minimum (0x01)
0x29, 0x02,        //   Usage Maximum (0x02)
0x75, 0x08,        //   Report Size (8)
0x95, 0x05,        //   Report Count (5)
0xB1, 0x02,        //   Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              // End Collection
```

I'm working on a new version where these options are customizable, I'll ask you again to test if you can!

carlossless, 2023-12-23:
@normano64 could you try the recently published 0.0.7? Should be enough to use the added `--part weikav-sugar65`.

normano64, 2023-12-23:
@carlossless It worked! 🥳

closed 2023-12-23 by carlossless