karstx/sinowisp #38: Royal Kludge RK61 rgb wired: write verification error?

timaxur, 2024-01-04:
Hello,

The tool can read and write the original firmware without reporting errors. But it reports an error on verification when writing a custom smk firmware:

~~~
> make flash
sinowealth-kb-tool write -p royalkludge-rk61-rgb-wired bin/main.hex
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00c7
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00c7
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Connected!
INFO  [sinowealth_kb_tool::isp] Erasing...
INFO  [sinowealth_kb_tool::isp] Writing...
INFO  [sinowealth_kb_tool::isp] Reading...
INFO  [sinowealth_kb_tool::isp] Verifying...
ERROR [sinowealth_kb_tool] Firmware Mismatch @ 0xeffb --- 0x02 != 0x00
make: *** [Makefile:59: flash] Error 1
~~~

If I then I use the tool to read the custom already flashed smk firmware from the keyboard and write it again then this verification error is not reported anymore... looks like the tool tries to write a read only portion and fails on verification? Anyway this doesn't seem to affect the programming, the smk custom firmware has been running happy. Let me know if you need more information about this or if this can be considered an expected no harmful behavior. 

Thank you!

carlossless, 2024-01-04:
@timaxur hm that's curious. You seem to have data in the last page of the firmware (specifically at `0xeffb`, like the log mentions). What version (commit hash) of SMK have you built here?

Due to a poor understanding of how the ISP bootloader worked, previous versions of SMK contained a `ljmp` instruction to main firmware at `0xeffb - 0xeffd`, but it's actually not necessary, [since the bootloader will take care of that itself](https://github.com/carlossless/sinowealth-kb-tool?tab=readme-ov-file#writing). That part of the firmware image should be kept blank though.

If you think there's a different issue, please share the hex file you're writing when you run into this issue and I'll look at it.

timaxur, 2024-01-05:
Thank you for the explanation, I'm going to close this issue because it doesn't cause harm and seems to be what you are describing: I'm using an SMK based on commit d65d1be and I see that the preboot.asm was removed a couple commits later.
I will post more details about the SMK [here](https://github.com/carlossless/smk/issues/3#issue-2057717439) shortly.

closed 2024-01-05 by timaxur