karstx/sinowisp #40: [device-report] Xinmeng XM-RF68

iegubkin, 2024-01-05:
## Device Info
- Model: Xinmeng XM-RF68
- Sinowealth Device: ?
- IC Label: K901+K3632 (printed in manual)
- Product Page: http://www.szxinmeng.com.cn/product/details/95 (non-working)

## Part Info

```
firmware_size: ?
vendor_id: 0x258a
product_id: 0x002a
bootloader_size: ?
page_size: ?
isp_usage_page: ?
isp_usage: ?
isp_index: ?
```

## Operations Tested

- [x] Read
- [ ] Write

## Platforms Tested

- [x] linux
- [ ] macos
- [ ] windows

Hello,
I have two of these keyboards. One working and one semi-working. I believe the MCUs to be similar/identical to the Royal Kludge RK61 and RK100. In fact, the reason that one of my keyboards is "semi-working" is that I managed to "successfully" flash firmware from the RK100 onto my XM-RF68 using Royal Kludge's Windows customization/upgrade tool. Surprisingly the damage was only to about 5 keys which no longer function as expected.

I was hoping that I would be able to use your program to dump the firmware from the working XM-RF68 and write it to the "semi-working" one (currently with RK100 firmware). 

The keyboards show up as "258a:002a SINO WEALTH Gaming KB" when plugged in.

Unfortunately the devices cannot be read using the following command:
```
# sinowealth-kb-tool read --vendor_id 0x258a --product_id 0x002a --firmware_size 61440 XM-RF68.hex

INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x002a
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x002a
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 10/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x002a
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
ERROR [sinowealth_kb_tool] Device not found
```
Thanks, Ivan.

iegubkin, 2024-01-05:
Tried changing index from 0, 1, and 2 with no success:
```
# sinowealth-kb-tool read --vendor_id 0x258a --product_id 0x002a --firmware_size 61440 --bootloader_size 4096 --page_size 2048 --isp_index 0 XM-RF68.hex

INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x002a
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
ERROR [sinowealth_kb_tool] Device not found
```
Maybe of use:
```
# lsusb --verbose -s 001:008

Bus 001 Device 008: ID 258a:002a SINO WEALTH Gaming KB 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x258a SINO WEALTH
  idProduct          0x002a Gaming KB 
  bcdDevice           10.01
  iManufacturer           1 SINO WEALTH
  iProduct                2 Gaming KB 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x003b
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      67
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     269
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```
Also:
```
# usbhid-dump --entity=all

001:008:001:DESCRIPTOR         1704472415.001375
 05 01 09 80 A1 01 85 01 19 81 29 83 15 00 25 01
 75 01 95 03 81 02 95 05 81 01 C0 05 0C 09 01 A1
 01 85 02 19 00 2A 3C 02 15 00 26 3C 02 95 01 75
 10 81 00 C0 06 00 FF 09 01 A1 01 85 05 15 00 26
 FF 00 19 01 29 02 75 08 95 05 B1 02 C0 05 01 09
 06 A1 01 85 06 05 07 19 04 29 70 15 00 25 01 75
 01 95 78 81 02 C0 06 00 FF 09 01 A1 01 85 09 15
 00 26 FF 00 09 00 75 08 96 F8 01 B1 02 C0 06 00
 FF 09 01 A1 01 85 0A 15 00 26 FF 00 09 00 75 08
 95 29 B1 02 C0 06 00 FF 09 01 A1 01 85 0B 15 00
 26 FF 00 09 00 75 08 95 7E B1 02 C0 05 01 09 02
 A1 01 85 0D 09 01 A1 00 05 09 15 00 25 01 19 01
 29 05 75 01 95 05 81 02 95 03 81 01 05 01 16 00
 80 26 FF 7F 09 30 09 31 75 10 95 02 81 06 15 81
 25 7F 09 38 75 08 95 01 81 06 05 0C 0A 38 02 95
 01 81 06 C0 C0 06 00 FF 09 01 A1 01 85 0C 15 00
 26 FF 00 09 00 75 08 96 80 07 B1 02 C0

001:008:000:DESCRIPTOR         1704472415.004138
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 95 08 75 01 81 02 95 01 75 08 81 03 95 06 75 08
 15 00 26 FF 00 05 07 19 00 2A FF 00 81 00 25 01
 95 05 75 01 05 08 19 01 29 05 91 02 95 01 75 03
 91 03 C0
```

iegubkin, 2024-01-05:
USB Descriptor and Request Parser
https://eleccelerator.com/usbdescreqparser/

001:008:001:DESCRIPTOR         1704472415.001375 --->
```
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x80,        // Usage (Sys Control)
0xA1, 0x01,        // Collection (Application)
0x85, 0x01,        //   Report ID (1)
0x19, 0x81,        //   Usage Minimum (Sys Power Down)
0x29, 0x83,        //   Usage Maximum (Sys Wake Up)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x01,        //   Logical Maximum (1)
0x75, 0x01,        //   Report Size (1)
0x95, 0x03,        //   Report Count (3)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x05,        //   Report Count (5)
0x81, 0x01,        //   Input (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection
0x05, 0x0C,        // Usage Page (Consumer)
0x09, 0x01,        // Usage (Consumer Control)
0xA1, 0x01,        // Collection (Application)
0x85, 0x02,        //   Report ID (2)
0x19, 0x00,        //   Usage Minimum (Unassigned)
0x2A, 0x3C, 0x02,  //   Usage Maximum (AC Format)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0x3C, 0x02,  //   Logical Maximum (572)
0x95, 0x01,        //   Report Count (1)
0x75, 0x10,        //   Report Size (16)
0x81, 0x00,        //   Input (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection
0x06, 0x00, 0xFF,  // Usage Page (Vendor Defined 0xFF00)
0x09, 0x01,        // Usage (0x01)
0xA1, 0x01,        // Collection (Application)
0x85, 0x05,        //   Report ID (5)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x19, 0x01,        //   Usage Minimum (0x01)
0x29, 0x02,        //   Usage Maximum (0x02)
0x75, 0x08,        //   Report Size (8)
0x95, 0x05,        //   Report Count (5)
0xB1, 0x02,        //   Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              // End Collection
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x06,        // Usage (Keyboard)
0xA1, 0x01,        // Collection (Application)
0x85, 0x06,        //   Report ID (6)
0x05, 0x07,        //   Usage Page (Kbrd/Keypad)
0x19, 0x04,        //   Usage Minimum (0x04)
0x29, 0x70,        //   Usage Maximum (0x70)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x01,        //   Logical Maximum (1)
0x75, 0x01,        //   Report Size (1)
0x95, 0x78,        //   Report Count (120)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection
0x06, 0x00, 0xFF,  // Usage Page (Vendor Defined 0xFF00)
0x09, 0x01,        // Usage (0x01)
0xA1, 0x01,        // Collection (Application)
0x85, 0x09,        //   Report ID (9)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x09, 0x00,        //   Usage (0x00)
0x75, 0x08,        //   Report Size (8)
0x96, 0xF8, 0x01,  //   Report Count (504)
0xB1, 0x02,        //   Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              // End Collection
0x06, 0x00, 0xFF,  // Usage Page (Vendor Defined 0xFF00)
0x09, 0x01,        // Usage (0x01)
0xA1, 0x01,        // Collection (Application)
[... 84 lines cut ...]
0x75, 0x08,        //   Report Size (8)
0x81, 0x03,        //   Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x06,        //   Report Count (6)
0x75, 0x08,        //   Report Size (8)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x05, 0x07,        //   Usage Page (Kbrd/Keypad)
0x19, 0x00,        //   Usage Minimum (0x00)
0x2A, 0xFF, 0x00,  //   Usage Maximum (0xFF)
0x81, 0x00,        //   Input (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x25, 0x01,        //   Logical Maximum (1)
0x95, 0x05,        //   Report Count (5)
0x75, 0x01,        //   Report Size (1)
0x05, 0x08,        //   Usage Page (LEDs)
0x19, 0x01,        //   Usage Minimum (Num Lock)
0x29, 0x05,        //   Usage Maximum (Kana)
0x91, 0x02,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x95, 0x01,        //   Report Count (1)
0x75, 0x03,        //   Report Size (3)
0x91, 0x03,        //   Output (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              // End Collection

// 67 bytes
// best guess: USB HID Report Descriptor
```

iegubkin, 2024-01-05:
So I tried on Windows and success with the "semi-working" (RK100 firmware) keyboard:
```
sinowealth-kb-tool.exe read --vendor_id 0x258a --product_id 0x002a --firmware_size 61440 xm-rf68.hex
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x002a
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x002a
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 3/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x002a
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 4/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x002a
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Connected!
INFO  [sinowealth_kb_tool::isp] Enabling firmware...
INFO  [sinowealth_kb_tool::isp] Reading...
INFO  [sinowealth_kb_tool] MD5: 93992856ff17c2f82748884ba66ed4a3
```
Moreover this triggered some changes that indicated a new usb device had been configured. The thought then came to my mind that I might be able to flash this keyboard with RK G68 firmware. Previously when I tried this, the firmware updater simply said "waiting ..." [actually  "waitting ..."]. So with nothing to lose, I again tried the Windows firmware update tool linked in this [reddit post](https://www.reddit.com/r/MechanicalKeyboards/comments/murcll/comment/h552qlf/) that perfectly matched the description of my keyboard. This time the keyboard was recognized and the firmware update process completed with a "Pass". My device is now recognized as "ID 258a:0052 SINO WEALTH Gaming KB" and the previously non-functioning keys are back working!

Thank you for your tool--it brought an ailing keyboard back to health! Let me know if I can provide any additional info which may be of use to your project.

iegubkin, 2024-01-05:
Unfortunately, the RK G68 firmware wasn't a great fit either. The RK G68 keyboard also has bluetooth, RF, and wired modes but they are actuated via a hardware switch. The XM RF-68 uses FN+ key combinations to change these modes. Also the shift was acting as an up arrow and the up arrow key was dead. On the plus side, the RK G68 firmware had home, end, insert keys unlike the stock XM RF-68 firmware.

So, back to my original plan: dump the firmware from the working XM RF-68 and write it to the ailing one (now with RK G68 firmware)

Dumping keyboard 1:
```
sinowealth-kb-tool.exe read --vendor_id 0x258a --product_id 0x002a --firmware_size 61440 xm-rf68.hex
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x002a
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x002a
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 3/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x002a
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 4/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x002a
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Connected!
INFO  [sinowealth_kb_tool::isp] Enabling firmware...
INFO  [sinowealth_kb_tool::isp] Reading...
INFO  [sinowealth_kb_tool] MD5: d580483b3e9f214ab96f65dcfb4ef795
```
Writing keyboard 2:
```
sinowealth-kb-tool.exe write --vendor_id 0x258a --product_id 0x0049 --firmware_size 61440 xm-rf68.hex
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Connected!
INFO  [sinowealth_kb_tool::isp] Erasing...
INFO  [sinowealth_kb_tool::isp] Writing...
INFO  [sinowealth_kb_tool::isp] Reading...
INFO  [sinowealth_kb_tool::isp] Verifying...
INFO  [sinowealth_kb_tool::isp] Enabling firmware...
```
And everything seems to work. Now, if it only had home, end, and insert keys ...
Thanks for this cool program!

carlossless, 2024-01-06:
@iegubkin Thanks for the report and I'm glad you got your keyboard back to a working state!

Could you please also share the MD5 checksum of the bootloader for these keyboards?

> IC Label: K901+K3632 (printed in manual)

Did you peak what's written on the IC inside the keyboard? The value you posted would suggest that this is a BYK901 which I don't know which actual sinowealth part this is (I guess it could be just another label for SH68F90). Would like to make sure that's the label before I try investigating this further.

Also, I'm surprised that the the tool didn't work for you on linux, but it did on windows. Did you make sure your user had the right permissions to interface with USB devices? https://github.com/carlossless/sinowealth-kb-tool?tab=readme-ov-file#linux

iegubkin, 2024-01-07:
Yes, why it's not working on my linux box is puzzling (but it fails a lot on Windows too).

I tried various usb ports including 2.0 ports directly attached to the motherboard. Still nothing. I was also running the script with root privileges.

Since Arch doesn't have the plugdev group, I combined your rule with the [wiki's advice](https://wiki.archlinux.org/title/udev#Allowing_regular_users_to_use_devices):

/etc/udev/rules.d/71-device-name.rules
SUBSYSTEMS=="usb", ATTRS{idVendor}=="258a", ATTRS{idProduct}=="002a", MODE="0660", TAG+="uaccess"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0603", ATTRS{idProduct}=="1020", MODE="0660", TAG+="uaccess"

Then I reloaded the udev rules:
udevadm control --reload
udevadm trigger

This seemed to allow a regular user to access the device but the result was the same--10 attempts ending with "Device not found".

Even on Windows this time it didn't want to respond. After quite a few attempts (and, a change of usb ports), I was able to pry out of it a response:
```
sinowealth-kb-tool.exe read --vendor_id 0x258a --product_id 0x002a --firmware_size 61440 --bootloader_size 4096 --page_size 2048 --isp_usage_page 0xff00 --isp_usage 0x0001 --isp_index 0 -b bootloader.hex
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x002a
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Connected!
INFO  [sinowealth_kb_tool::isp] Enabling firmware...
INFO  [sinowealth_kb_tool::isp] Reading...
INFO  [sinowealth_kb_tool] MD5: 2d169670eae0d36eae8188562c1f66e8
```
As for the IC, I removed the board from the case and it revealed a SH68F90U-00999 chip.
[image]
Let me know if I can be of further assistance. Thanks again for your efforts!

carlossless, 2024-01-13:
@iegubkin I believe I figured out what the issue on linux is (#43). Please try running https://github.com/carlossless/sinowealth-kb-tool/releases/tag/v0.0.8 it should work now.

iegubkin, 2024-01-13:
Great job! The new version worked perfectly, first time as a regular user (using the tweaked udev rule for Arch I mentioned above). Let me know if you still would like any of these firmware files.

-- firmware dump excluding isp bootloader
```
$ ./sinowealth-kb-tool read --vendor_id 0x258a --product_id 0x002a --firmware_size 61440 xm-rf68.hex
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x002a
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x002a
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Connected!
INFO  [sinowealth_kb_tool::isp] Enabling firmware...
INFO  [sinowealth_kb_tool::isp] Reading...
INFO  [sinowealth_kb_tool] MD5: d580483b3e9f214ab96f65dcfb4ef795
```

--full firmware dump including bootloader
```
$ ./sinowealth-kb-tool read --vendor_id 0x258a --product_id 0x002a --firmware_size 61440 --full xm-rf68_full.hex
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x002a
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Connected!
INFO  [sinowealth_kb_tool::isp] Enabling firmware...
INFO  [sinowealth_kb_tool::isp] Reading...
INFO  [sinowealth_kb_tool] MD5: 860c97b69f324fc6a5151a8bdaec7574
```

carlossless, 2024-01-13:
@iegubkin thanks for testing it again! glad to see it works!

> Let me know if you still would like any of these firmware files.

I think I don't need them right now. Perhaps I reach out in the future!

> Now, if it only had home, end, and insert keys ...

Btw, I am not sure if you needed physical keys here or just a layout in firmware that would make them accessible through a key combination, but if it's the latter you might want to check out https://github.com/carlossless/smk.

closed 2024-01-13 by carlossless