karstx/sinowisp #45: [device-report] royal kludge rk84 iso layout (ger)

ViktorTrojan, 2024-02-08:
## Device Info

- Sinowealth Device: _example `SH68F90A`_
- IC Label: _example `BYK916`_
- Product Page: https://rkgamingstore.com/products/rk84-75-percent-keyboard?variant=43341644857565

## Part Info

```
firmware_size: 61440
vendor_id: 0x258A
product_id: 0xF4
bootloader_size: 4096 # necessary if not default
page_size: 2048 # necessary if not default
isp_usage_page: 0xff00 # necessary if not default
isp_usage: 0x0001 # necessary if not default
isp_index: 0 # necessary if not default
```

## Operations Tested

> Is firmware R/W ment? if so, no.

- [ ] Read
- [ ] Write

## Platforms Tested

- [ ] linux
- [ ] macos
- [ ] windows

## Checksums

- Bootloader MD5: `beefcafebeefcafebeefcafebeefcafe`
- Stock Firmware MD5: `deadbeefdeadbeefdeadbeefdeadbeef`

## HID Dump

A dump from [usbhid-dump](https://github.com/DIGImend/usbhid-dump), [win-hid-dump](https://github.com/todbot/win-hid-dump) or [mac-hid-dump](https://github.com/todbot/mac-hid-dump)

<details>
<summary>HID Tool Output</summary>

```
# RK84 ISO Layout using win-hid-dump

258A:00F4: SINO WEALTH - RK Bluetooth Keyboard
PATH:\\?\hid#vid_258a&pid_00f4&mi_01&col03#9&1c1eb664&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  06  00  FF  09  01  A1  01  85  05  15  00  25  01  35  00  45
  01  65  00  55  00  75  01  95  28  B1  03  C1  00
  (29 bytes)
258A:00F4: SINO WEALTH - RK Bluetooth Keyboard
PATH:\\?\hid#vid_258a&pid_00f4&mi_01&col05#9&1c1eb664&0&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  06  00  FF  09  01  A1  01  85  0A  09  00  15  00  25  FF  35
  00  45  00  65  00  55  00  75  08  95  40  B1  02  C1  00
  (31 bytes)
258A:00F4: SINO WEALTH - RK Bluetooth Keyboard
PATH:\\?\hid#vid_258a&pid_00f4&mi_01&col04#9&1c1eb664&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
DESCRIPTOR:
  05  01  09  06  A1  01  85  06  05  07  19  04  29  70  15  00
  25  01  35  00  45  01  65  00  55  00  75  01  95  6D  81  02
  95  0B  81  03  C1  00
  (38 bytes)
258A:00F4: SINO WEALTH - RK Bluetooth Keyboard
PATH:\\?\hid#vid_258a&pid_00f4&mi_00#9&8ee9a26&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
DESCRIPTOR:
  05  01  09  06  A1  01  05  07  19  E0  29  E7  15  00  25  01
  35  00  45  01  65  00  55  00  75  01  95  08  81  02  95  38
  81  03  05  08  19  01  29  05  95  05  91  02  95  03  91  03
  C1  00
  (50 bytes)
258A:00F4: SINO WEALTH - RK Bluetooth Keyboard
PATH:\\?\hid#vid_258a&pid_00f4&mi_01&col02#9&1c1eb664&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  05  0C  09  01  A1  01  85  02  15  00  25  01  35  00  45  01
  65  00  55  00  75  01  95  10  81  03  C1  00
  (28 bytes)
258A:00F4: SINO WEALTH - RK Bluetooth Keyboard
PATH:\\?\hid#vid_258a&pid_00f4&mi_01&col01#9&1c1eb664&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  05  01  09  80  A1  01  85  01  19  81  29  83  15  00  25  01
  35  00  45  01  65  00  55  00  75  01  95  03  81  02  95  05
  81  03  C1  00
  (36 bytes)
```

</details>

carlossless, 2024-02-09:
Hey @ViktorTrojan, have you tried running the tool for your device? Since the device it not added you can try interfacing with it with all the custom parameters. It should be something like this:

ViktorTrojan, 2024-02-17:
When I try to run it I get an error. What can I do to debug this? I have the firmware update software from the manufacturer if that helps.

sinowealth-kb-tool.exe read  --vendor_id 0x258a --product_id 0x00f4 --firmware_size 61440 foobar.hex
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00f4
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00f4
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 3/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00f4
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Connected!
INFO  [sinowealth_kb_tool::isp] Enabling firmware...
ERROR [sinowealth_kb_tool] hidapi error: HidD_SetFeature: (0x00000001) Incorrect function.

carlossless, 2024-02-17:
```
ERROR [sinowealth_kb_tool] hidapi error: HidD_SetFeature: (0x00000001) Incorrect function.
```

Hm, that's interesting... Have you tried running it only once? Try running it again.

If this issue persists, I assume this device might be running a different bootloader than the other keyboards encountered so far.

Luro02, 2024-02-28:
I have the same keyboard and am running into the same issue. I recorded the communication with wireshark when flashing the firmware with the official application:
[firmware_flash.zip](https://github.com/carlossless/sinowealth-kb-tool/files/14435242/firmware_flash.zip)

The record starts with me connecting the keyboard to the pc and then executing the firmware update without pressing any keys.

Luro02, 2024-02-29:
I am currently trying to dump the keyboard firmware with https://github.com/gashtaan/sinowealth-8051-dumper, but so far I wasn't able to.

Most of the time I get a "Connection failed", except for one time where is asked me to press enter and then printed `FF` bytes for everything, so I guess that didn't work either.

Luro02, 2024-02-29:
> I am currently trying to dump the keyboard firmware with https://github.com/gashtaan/sinowealth-8051-dumper, but so far I wasn't able to.
>
> Most of the time I get a "Connection failed", except for [...]

Here is the full dump:
<details>
  <summary>RK84 firmware</summary>

```
SinoWealth 8051-based MCU flash dumper
Waiting for Vref to get high - enable power to target manually:
Vref is now high - resuming
Connection established

Press ENTER to proceed

JTAG ID: F690

Dumping part number:
68F90A0000

Dumping code options:
000000007CDCA4E063C0300030003000
30000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
6E2E2805220621151519054002300100
0F000088000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000

Dumping flash memory:
02F00002308641085F000002B02A758F
01222202B0300250E7220002B0369003
9DE004F0E0D39402400FE4F090039CE0
04F0E0947E4002E4F0220002A853D249
22000002A810900898EFF07E01300406
900943E52CF0900898E0FF643F600302
00F9900943E030E179EFFD7C00E4FF12
9178EF24FCFFE43403FE900898E0FD75
F002A42493F582E5F03400F583EEF0A3
EFF07F01129178EF24FCFFE43403FE90
0898E0FD75F002A4248FF582E5F03401
F583EEF0A3EFF07F02129178EF24FCFF
E43403FE900898E075F002A4248BF582
E5F03402F583EEF0A3EFF0D23FE4FE02
[... 4061 lines cut ...]
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
645A702CE5F064A57026C2AF758170F5
B112FADAE4F591F592F593F594F595F5
96F5B17FD07E0712FBA5E4F5B112FC03
02FC2B00000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000

Done!
```
</details>

carlossless, 2024-02-29:
@Luro02 thanks for providing the dumps. Although, as you can probably tell, the dump you got from [sinowealth-8051-dumper](https://github.com/gashtaan/sinowealth-8051-dumper) has sectors of your main firmware blanked out.

If you're interested in getting the firmware, then that doesn't matter, because the USB packets you captured through wireshark have the full firmware payload, including the bytes that should be in those blanked-out sectors.

The bootloader part of that dump is fully intact and I checked that it's another instance of `cfc8661da8c9d7e351b36c0a763426aa`, which makes me question why the tool failed at the firmware enable step.

* I'm guessing you tried reading the firmware? Did you try writing anything?
* Are you running on windows machine?
* Could you try running the tool on a linux machine?

Also, there's something I noticed in your wireshark dump: does the update utility automatically reset your keyboard from the bootloader back to the newly flashed main firmware?

Luro02, 2024-03-01:
> Although, as you can probably tell, the dump you got from [sinowealth-8051-dumper](https://github.com/gashtaan/sinowealth-8051-dumper) has sectors of your main firmware blanked out.

Oh, I assumed that was just padding.

> I'm guessing you tried reading the firmware?
Are you running on windows machine?
Could you try running the tool on a linux machine?

I am running the tool on windows. I will try running it on Linux.

> Did you try writing anything?

I did not try writing to it, because I do not want to brick my keyboard. Based on the code, this would not work, because it fails here
https://github.com/carlossless/sinowealth-kb-tool/blob/0ff0965d0c6aaa932414621002680142a65ae61d/src/isp.rs#L69

which is executed independently of the read/write choice.

And like mentioned above, the actual failure is here
https://github.com/carlossless/sinowealth-kb-tool/blob/0ff0965d0c6aaa932414621002680142a65ae61d/src/isp.rs#L241-L245

because the keyboard doesn't understand the command?

> does the update utility automatically reset your keyboard from the bootloader back to the newly flashed main firmware?

I think it does. I have recorded a video of the flashing process:

https://github.com/carlossless/sinowealth-kb-tool/assets/24826124/f9c1ed69-2cd4-4b72-9007-43249c9036d9

Before the video was taken it had the rainbow effect that you can see at the end of the video, then it locks the colors in the pattern you can see at the start of the video. The colors are random, but I always have this pattern.

It resets all settings I have made (but I guess that is obvious based on the communication I recorded).

Luro02, 2024-03-01:
This is what I got on WSL:
```
RUST_APP_LOG=TRACE ./sinowealth-kb-tool read --vendor_id 0x258a --product_id 0x00f4 --firmware_size 61440 foobar.hex
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00f4
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00f4
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 3/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00f4
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 4/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00f4
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 5/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00f4
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 6/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00f4
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 7/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00f4
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 8/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00f4
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 9/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00f4
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 10/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00f4
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
ERROR [sinowealth_kb_tool] Device not found
```

I have cloned the repo and made some changes to have more output, for some reason it did work then???
I have to investigate this further, but this is my current output:

<details open>
  <summary>Click me</summary>

```cmd
> cargo run -- read --vendor_id 0x258a --product_id 0x00f4 --firmware_size 61440 foobar.hex
   Compiling sinowealth-kb-tool v0.0.8 (D:\projects\sinowealth-kb-tool)
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.82s
     Running `target\debug\sinowealth-kb-tool.exe read --vendor_id 0x258a --product_id 0x00f4 --firmware_size 61440 foobar.hex`
DEBUG [sinowealth_kb_tool::isp] Listing all connected devices...
DEBUG [sinowealth_kb_tool::isp] Found Device(vid=0x0db0, pid=0x0076): manufacturer=Some("MSI") product=Some("MYSTIC LIGHT ") path="\\\\?\\HID#VID_0DB0&PID_0076#9&37d31a3f&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}" usage_page=0x0001 usage=0x0000
DEBUG [sinowealth_kb_tool::isp] Found Device(vid=0x04d9, pid=0xa09f): manufacturer=Some("E-Signal") product=Some("USB Gaming Mouse") path="\\\\?\\HID#VID_04D9&PID_A09F&MI_00#8&802cee6&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}" usage_page=0x0001 usage=0x0002
DEBUG [sinowealth_kb_tool::isp] Found Device(vid=0x04d9, pid=0xa09f): manufacturer=Some("E-Signal") product=Some("USB Gaming Mouse") path="\\\\?\\HID#VID_04D9&PID_A09F&MI_01&Col01#8&2bda0ca8&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\\KBD" usage_page=0x0001 usage=0x0006
DEBUG [sinowealth_kb_tool::isp] Found Device(vid=0x04d9, pid=0xa09f): manufacturer=Some("E-Signal") product=Some("USB Gaming Mouse") path="\\\\?\\HID#VID_04D9&PID_A09F&MI_01&Col02#8&2bda0ca8&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}" usage_page=0x0001 usage=0x0080
DEBUG [sinowealth_kb_tool::isp] Found Device(vid=0x04d9, pid=0xa09f): manufacturer=Some("E-Signal") product=Some("USB Gaming Mouse") path="\\\\?\\HID#VID_04D9&PID_A09F&MI_01&Col03#8&2bda0ca8&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}" usage_page=0x000c usage=0x0001
DEBUG [sinowealth_kb_tool::isp] Found Device(vid=0x04d9, pid=0xa09f): manufacturer=Some("E-Signal") product=Some("USB Gaming Mouse") path="\\\\?\\HID#VID_04D9&PID_A09F&MI_01&Col04#8&2bda0ca8&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}" usage_page=0xff01 usage=0xff01
DEBUG [sinowealth_kb_tool::isp] Found Device(vid=0x04d9, pid=0xa09f): manufacturer=Some("E-Signal") product=Some("USB Gaming Mouse") path="\\\\?\\HID#VID_04D9&PID_A09F&MI_02#8&39192188&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}" usage_page=0xff00 usage=0xff00
[... cut ...]
DEBUG [sinowealth_kb_tool::isp] Reading page 23 @ offset 0xb800
DEBUG [sinowealth_kb_tool::isp] Reading page 24 @ offset 0xc000
DEBUG [sinowealth_kb_tool::isp] Reading page 25 @ offset 0xc800
DEBUG [sinowealth_kb_tool::isp] Reading page 26 @ offset 0xd000
DEBUG [sinowealth_kb_tool::isp] Reading page 27 @ offset 0xd800
DEBUG [sinowealth_kb_tool::isp] Reading page 28 @ offset 0xe000
DEBUG [sinowealth_kb_tool::isp] Reading page 29 @ offset 0xe800
INFO  [sinowealth_kb_tool] MD5: 7658c72d2cdc04ab66f1216a206f6913
```
</details>

Luro02, 2024-03-01:
For some reason, the tool is now working with my keyboard:
<details open>
  <summary>Output</summary>

```cmd
   Compiling sinowealth-kb-tool v0.0.8 (D:\projects\sinowealth-kb-tool)
    Finished dev [unoptimized + debuginfo] target(s) in 0.68s
     Running `target\debug\sinowealth-kb-tool.exe read --vendor_id 0x258a --product_id 0x00f4 --firmware_size 61440 foobar.hex`
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00f4
DEBUG [sinowealth_kb_tool::isp] Found Device: "\\\\?\\HID#VID_258A&PID_00F4&MI_01&Col03#9&f505e63&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}" 0xff00 0x0001
DEBUG [sinowealth_kb_tool::isp] Opening: "\\\\?\\HID#VID_258A&PID_00F4&MI_01&Col03#9&f505e63&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}"
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00f4
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 3/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00f4
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
DEBUG [sinowealth_kb_tool::isp] Found Device: "\\\\?\\HID#VID_0603&PID_1020&Col02#8&2da0b3f2&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}" 0xff00 0x0001
DEBUG [sinowealth_kb_tool::isp] Found Device: "\\\\?\\HID#VID_0603&PID_1020&Col03#8&2da0b3f2&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}" 0xff00 0x0001
DEBUG [sinowealth_kb_tool::isp] Request device: "\\\\?\\HID#VID_0603&PID_1020&Col02#8&2da0b3f2&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}"
DEBUG [sinowealth_kb_tool::isp] Data device: "\\\\?\\HID#VID_0603&PID_1020&Col03#8&2da0b3f2&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}"
INFO  [sinowealth_kb_tool::isp] Connected!
INFO  [sinowealth_kb_tool::isp] Enabling firmware...
INFO  [sinowealth_kb_tool::isp] Reading...
DEBUG [sinowealth_kb_tool::isp] Reading page 0 @ offset 0x0000
DEBUG [sinowealth_kb_tool::isp] Reading page 1 @ offset 0x0800
DEBUG [sinowealth_kb_tool::isp] Reading page 2 @ offset 0x1000
DEBUG [sinowealth_kb_tool::isp] Reading page 3 @ offset 0x1800
DEBUG [sinowealth_kb_tool::isp] Reading page 4 @ offset 0x2000
DEBUG [sinowealth_kb_tool::isp] Reading page 5 @ offset 0x2800
DEBUG [sinowealth_kb_tool::isp] Reading page 6 @ offset 0x3000
DEBUG [sinowealth_kb_tool::isp] Reading page 7 @ offset 0x3800
DEBUG [sinowealth_kb_tool::isp] Reading page 8 @ offset 0x4000
DEBUG [sinowealth_kb_tool::isp] Reading page 9 @ offset 0x4800
DEBUG [sinowealth_kb_tool::isp] Reading page 10 @ offset 0x5000
DEBUG [sinowealth_kb_tool::isp] Reading page 11 @ offset 0x5800
DEBUG [sinowealth_kb_tool::isp] Reading page 12 @ offset 0x6000
DEBUG [sinowealth_kb_tool::isp] Reading page 13 @ offset 0x6800
DEBUG [sinowealth_kb_tool::isp] Reading page 14 @ offset 0x7000
DEBUG [sinowealth_kb_tool::isp] Reading page 15 @ offset 0x7800
DEBUG [sinowealth_kb_tool::isp] Reading page 16 @ offset 0x8000
DEBUG [sinowealth_kb_tool::isp] Reading page 17 @ offset 0x8800
DEBUG [sinowealth_kb_tool::isp] Reading page 18 @ offset 0x9000
DEBUG [sinowealth_kb_tool::isp] Reading page 19 @ offset 0x9800
DEBUG [sinowealth_kb_tool::isp] Reading page 20 @ offset 0xa000
DEBUG [sinowealth_kb_tool::isp] Reading page 21 @ offset 0xa800
DEBUG [sinowealth_kb_tool::isp] Reading page 22 @ offset 0xb000
DEBUG [sinowealth_kb_tool::isp] Reading page 23 @ offset 0xb800
DEBUG [sinowealth_kb_tool::isp] Reading page 24 @ offset 0xc000
DEBUG [sinowealth_kb_tool::isp] Reading page 25 @ offset 0xc800
DEBUG [sinowealth_kb_tool::isp] Reading page 26 @ offset 0xd000
DEBUG [sinowealth_kb_tool::isp] Reading page 27 @ offset 0xd800
DEBUG [sinowealth_kb_tool::isp] Reading page 28 @ offset 0xe000
DEBUG [sinowealth_kb_tool::isp] Reading page 29 @ offset 0xe800
INFO  [sinowealth_kb_tool] MD5: 7658c72d2cdc04ab66f1216a206f6913
```

</details>

The only problem is that the tool leaves the keyboard in the bootloader mode, which could be perceived as a bricked keyboard:
[image]
and of course the keyboard tool where you could flash the firmware does not detect it:
[image]

Luckily, the real flashing tool does not care about that and works, even when the keyboard is in this state (unplugging the keyboard works as well...).

Writing works but leaves the keyboard in bootloader mode too.

### For future reference in case someone else runs into this:

The RK Keyboard does a web request checking for updates, which you can intercept with for example fiddler classic. The request is currently done at the start of the application and when you press the "Firmware Update" button. (This is the link: http://www.rkgaming.com/Work/Update-RGB/update_version.xml)

In there you can find the link to download the executable for your keyboard. That is the one executed by the rk software as well.

carlossless, 2024-03-02:
> I did not try writing to it, because I do not want to brick my keyboard. Based on the code, this would not work, because it fails here which is executed independently of the read/write choice.

Then you've been experiencing a different issue from @ViktorTrojan. They are experiencing problems that can occur only after that point. (just clarifying this, since you mentioned that it's the same issue in your [first comment](https://github.com/carlossless/sinowealth-kb-tool/issues/45#issuecomment-1969100960)).

> I think it does. I have recorded a video of the flashing process

> The only problem is that the tool leaves the keyboard in the bootloader mode, which could be perceived as a bricked keyboard

I wasn't aware that there are bootloader versions that have a built-in reset command (mine does not have it). I'll test out `cfc8661da8c9d7e351b36c0a763426aa` on my hardware and add that feature into this tool for parts/devices that can support it ~ thanks!

> For some reason, the tool is now working with my keyboard

Hm, that's peculiar. So you're saying in the end it started working without doing any changes to the tool or your environment?

Luro02, 2024-03-02:
> So you're saying in the end it started working without doing any changes to the tool or your environment?

Yes

Luro02, 2024-03-03:
> > So you're saying in the end it started working without doing any changes to the tool or your environment?
> 
> Yes

Today I was unable to read/write the keyboard again, but thanks to all the logs I posted, I found the culprit:

In the logs that I posted where it flashed successfully it selects the request and data device like this:
```
DEBUG [sinowealth_kb_tool::isp] Request device: "\\\\?\\HID#VID_0603&PID_1020&Col02#8&2da0b3f2&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}"
DEBUG [sinowealth_kb_tool::isp] Data device: "\\\\?\\HID#VID_0603&PID_1020&Col03#8&2da0b3f2&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}"
```
when it failed to read/flash with the message mentioned, it selected the devices like this
```
DEBUG [sinowealth_kb_tool::isp] Request device: "\\\\?\\HID#VID_0603&PID_1020&Col03#8&2da0b3f2&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}" 0xff00 0x0001
DEBUG [sinowealth_kb_tool::isp] Data device: "\\\\?\\HID#VID_0603&PID_1020&Col02#8&2da0b3f2&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}" 0xff00 0x0001
```
> Then you've been experiencing a different issue from @ViktorTrojan. They are experiencing problems that can occur only after that point. (just clarifying this, since you mentioned that it's the same issue in your https://github.com/carlossless/sinowealth-kb-tool/issues/45#issuecomment-1969100960).

with this device selection I get the same error:
```
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00f4
DEBUG [sinowealth_kb_tool::isp] Found Device: "\\\\?\\HID#VID_258A&PID_00F4&MI_01&Col03#9&f505e63&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}" 0xff00 0x0001
DEBUG [sinowealth_kb_tool::isp] Opening: "\\\\?\\HID#VID_258A&PID_00F4&MI_01&Col03#9&f505e63&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}"
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00f4
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 3/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x00f4
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
DEBUG [sinowealth_kb_tool::isp] Found Device: "\\\\?\\HID#VID_0603&PID_1020&Col03#8&2da0b3f2&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}" 0xff00 0x0001
DEBUG [sinowealth_kb_tool::isp] Found Device: "\\\\?\\HID#VID_0603&PID_1020&Col02#8&2da0b3f2&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}" 0xff00 0x0001
DEBUG [sinowealth_kb_tool::isp] Request device: "\\\\?\\HID#VID_0603&PID_1020&Col03#8&2da0b3f2&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}"
DEBUG [sinowealth_kb_tool::isp] Data device: "\\\\?\\HID#VID_0603&PID_1020&Col02#8&2da0b3f2&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}"
INFO  [sinowealth_kb_tool::isp] Connected!
INFO  [sinowealth_kb_tool::isp] Erasing...
ERROR [sinowealth_kb_tool] hidapi error: HidD_SetFeature: (0x00000001) Incorrect function.
```

The tool selects the wrong device for request/data. It should select them based on the path, so the one with `&0001` is the request device and the other one (with `&0002`) is the data device. (The paths differ in the `&Col03` vs `&Col02` as well).

carlossless, 2024-03-09:
@ViktorTrojan the latest [v0.0.9](https://github.com/carlossless/sinowealth-kb-tool/releases/tag/v0.0.9) release, should likely fix solve the issue you've been experiencing. Please try it out and let us know.

ViktorTrojan, 2024-04-13:
> @ViktorTrojan the latest [v0.0.9](https://github.com/carlossless/sinowealth-kb-tool/releases/tag/v0.0.9) release should likely solve the issue you've been experiencing. Please try it out and let us know.

Sorry for late respone, tried it with latest version (v.0.0.10) which worked

closed 2024-04-13 by ViktorTrojan