karstx/sinowisp #50: [device-report] Redragon Shaco Pro (k641 pro)

Danial-Movahed, 2024-03-17:
## Device Info

- Sinowealth Device: `SH68F90A`
- IC Label: `BYK916`
- Product Page: https://redragonshop.com/products/shaco-k641-aluminum-wireless-keyboard

## Part Info

```
firmware_size: 61440
vendor_id: 0x258a
product_id: 0x0049
bootloader_size: 4096 # necessary if not default
page_size: 2048 # necessary if not default
isp_usage_page: 0xff00 # necessary if not default
isp_usage: 0x0001 # necessary if not default
isp_index: 0 # necessary if not default
```

## Operations Tested

- [x] Read
- [ ] Write

## Platforms Tested

- [x] linux
- [ ] macos
- [ ] windows

## Checksums

- Bootloader MD5: `3e0ebd0c440af5236d7ff8872343f85d`
- Stock Firmware MD5: `c506173c042656d078dbaf54906ac44b`

## HID Dump

A dump from [usbhid-dump](https://github.com/DIGImend/usbhid-dump), [win-hid-dump](https://github.com/todbot/win-hid-dump) or [mac-hid-dump](https://github.com/todbot/mac-hid-dump)

<details>
<summary>HID Tool Output</summary>

```
# Redragon Shaco Pro using hid-dump
❯ sudo usbhid-dump -m 258a:0049
001:035:001:DESCRIPTOR         1710647674.196170
 06 01 00 09 80 A1 01 85 01 19 81 29 83 15 00 25
 01 95 03 75 01 81 02 95 01 75 05 81 01 C0 05 0C
 09 01 A1 01 85 02 19 00 2A FF 02 15 00 26 FF 7F
 95 01 75 10 81 00 C0 06 00 FF 09 01 A1 01 85 03
 15 00 26 FF 00 09 2F 75 08 95 03 81 02 C0 05 01
 09 06 A1 01 85 04 05 07 19 04 29 70 15 00 25 01
 75 01 95 78 81 02 C0 06 00 FF 09 01 A1 01 85 05
 15 00 26 FF 00 19 01 29 02 75 08 95 05 B1 02 C0
 06 00 FF 09 01 A1 01 85 06 15 00 26 FF 00 19 01
 29 02 75 08 96 07 04 B1 02 C0 05 01 09 02 A1 01
 85 07 09 01 A1 00 05 09 15 00 25 01 19 01 29 05
 75 01 95 05 81 02 95 03 81 01 05 01 16 00 80 26
 FF 7F 09 30 09 31 75 10 95 02 81 06 15 81 25 7F
 09 38 75 08 95 01 81 06 05 0C 0A 38 02 95 01 81
 06 C0 C0 06 00 FF 09 01 A1 01 85 08 15 00 26 FF
 00 09 00 75 08 96 7D 01 B1 02 C0

001:035:000:DESCRIPTOR         1710647674.199069
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 95 08 75 01 81 02 95 01 75 08 81 03 95 06 75 08
 15 00 26 FF 00 05 07 19 00 2A FF 00 81 00 25 01
 95 05 75 01 05 08 19 01 29 05 91 02 95 01 75 03
 91 03 C0
```

</details>
Hi. I'm trying to read the stock firmware and bootloader of this keyboard with

```
sinowealth-kb-tool-bin read --full \
    --vendor_id 0x258a \
    --product_id 0x0049 \
    --firmware_size 61440 \
    --bootloader_size 4096 \
    --page_size 2048 \
    --reboot false \
    StockFW.hex
```

Most of the times it will give this error and leaves the keyboard in the bootloader mode with random rgb lights:
<details>
<summary>Error log</summary>

```
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 3/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
thread 'main' panicked at src/isp.rs:242:64:
called `Result::unwrap()` on an `Err` value: HidApiError { message: "hid_error is not implemented yet" }
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```
</details>
The keyboard won't work until re plugged.

But sometimes it will work just fine.
<details>
<summary>Successful read log</summary>

```
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 3/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 4/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Connected!
INFO  [sinowealth_kb_tool::isp] Enabling firmware...
INFO  [sinowealth_kb_tool::isp] Reading...
INFO  [sinowealth_kb_tool] MD5: f885cbb3a33602ae52bae90d491e1303
```
</details>

carlossless, 2024-03-18:
@Danial-Movahed thanks a lot for your report and glad to hear you managed to fix your keyboard!

Based on the output from `usbhid-dump` and all other redragon keyboards reported so far, the isp_index should be 1, but this would only affect windows users. Would you be able to test this on a windows device and confirm whether isp_index=1 is the option that works for this keyboard?

SIDE-NOTE: redragon uses the same pId (and naturally, the same vId) for all of their keyboards, or at least it seems that way :(

Danial-Movahed, 2024-03-19:
Ok, Here you go. Dump from win-hiddump on windows 11 22H2.

<details>
<summary>HID Tool Output</summary>

```
# Redragon Shaco Pro using winhiddump
258A:0049: BY Tech - Gaming Keyboard
PATH:\\?\hid#vid_258a&pid_0049&mi_00#7&30a43d12&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
DESCRIPTOR:
  05  01  09  06  A1  01  05  07  19  E0  29  E7  15  00  25  01
  35  00  45  01  65  00  55  00  75  01  95  08  81  02  95  38
  81  03  05  08  19  01  29  05  95  05  91  02  95  03  91  03
  C1  00
  (50 bytes)
258A:0049: BY Tech - Gaming Keyboard
PATH:\\?\hid#vid_258a&pid_0049&mi_01&col01#7&cccff50&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  05  01  09  80  A1  01  85  01  19  81  29  83  15  00  25  01
  35  00  45  01  65  00  55  00  75  01  95  03  81  02  95  05
  81  03  C1  00
  (36 bytes)
258A:0049: BY Tech - Gaming Keyboard
PATH:\\?\hid#vid_258a&pid_0049&mi_01&col02#7&cccff50&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  05  0C  09  01  A1  01  85  02  15  00  25  01  35  00  45  01
  65  00  55  00  75  01  95  10  81  03  C1  00
  (28 bytes)
258A:0049: BY Tech - Gaming Keyboard
PATH:\\?\hid#vid_258a&pid_0049&mi_01&col03#7&cccff50&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  06  00  FF  09  01  A1  01  85  03  09  2F  15  00  25  FF  35
  00  45  00  65  00  55  00  75  08  95  03  81  02  C1  00
  (31 bytes)
258A:0049: BY Tech - Gaming Keyboard
PATH:\\?\hid#vid_258a&pid_0049&mi_01&col05#7&cccff50&0&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  06  00  FF  09  01  A1  01  85  05  15  00  25  01  35  00  45
  01  65  00  55  00  75  01  95  28  B1  03  C1  00
  (29 bytes)
258A:0049: BY Tech - Gaming Keyboard
PATH:\\?\hid#vid_258a&pid_0049&mi_01&col04#7&cccff50&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
DESCRIPTOR:
  05  01  09  06  A1  01  85  04  05  07  19  04  29  70  15  00
  25  01  35  00  45  01  65  00  55  00  75  01  95  6D  81  02
  95  0B  81  03  C1  00
  (38 bytes)
258A:0049: BY Tech - Gaming Keyboard
PATH:\\?\hid#vid_258a&pid_0049&mi_01&col08#7&cccff50&0&0007#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  06  00  FF  09  01  A1  01  85  08  09  00  15  00  25  FF  35
  00  45  00  65  00  55  00  75  08  96  7D  01  B1  02  C1  00

  (32 bytes)
258A:0049: BY Tech - Gaming Keyboard
PATH:\\?\hid#vid_258a&pid_0049&mi_01&col06#7&cccff50&0&0005#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  06  00  FF  09  01  A1  01  85  06  15  00  25  01  35  00  45
  01  65  00  55  00  75  01  96  38  20  B1  03  C1  00
  (30 bytes)
258A:0049: BY Tech - Gaming Keyboard
PATH:\\?\hid#vid_258a&pid_0049&mi_01&col07#7&cccff50&0&0006#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  05  01  09  02  A1  01  85  07  09  01  A1  00  05  09  19  01
  29  05  15  00  25  01  35  00  45  01  65  00  55  00  75  01
  95  05  81  02  95  03  81  03  05  01  09  30  26  FF  7F  45
  00  75  10  95  01  81  06  09  31  81  06  09  38  25  7F  75
  08  81  06  05  0C  0A  38  02  81  06  C1  00  C1  00
  (78 bytes)
```

</details>

I also have a hackintosh machine, If you want I can get a dump from there and test this tool.

btw when I bricked my keyboard, I took it to the seller's warranty service because I'm in Iran, there is no official warranty service for redragon and they told me that to fix this the chip must be replaced and after that it needs to be programmed which the programmer and the firmware file is redragon confidential therefore we can't fix this. And they wanted me to pay one third of the price of keyboard to give me a new one XD. But thanks to your tools I was able to fix it without spending any money.

Danial-Movahed, 2024-03-19:
I tested read with isp_index 0 and 1 on windows. Only with isp_index 1 it works. Here is the log:
<details><summary>isp_index 0</summary>

```
.\sinowealth-kb-tool read --full --vendor_id 0x258a --product_id 0x0049 --firmware_size 61440  --bootloader_size 4096 --reboot false --isp_index 0 StockFW.hex
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 3/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 4/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 5/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 6/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 7/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 8/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 9/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 10/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
ERROR [sinowealth_kb_tool] Device not found
```

</details>
<details><summary>isp_index 1</summary>

```
.\sinowealth-kb-tool read --full --vendor_id 0x258a --product_id 0x0049 --firmware_size 61440  --bootloader_size 4096 --reboot false --isp_index 1 StockFW.hex
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 3/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 4/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Connected!
INFO  [sinowealth_kb_tool::isp] Enabling firmware...
INFO  [sinowealth_kb_tool::isp] Reading...
INFO  [sinowealth_kb_tool] MD5: bfd63f71d4838ea206e80516ff5f05ab
```

</details>

carlossless, 2024-03-22:
@Danial-Movahed great, thanks!

closed 2024-03-22 by carlossless