karstx/sinowisp #58: [device-report] Machenike K500-B61

wyrla, 2024-03-28:
## Device Info

- Sinowealth Device: _example `SH68F90A`_
- IC Label: `BYK916`
- Product Page: https://global.machenike.com/products/k500-b61
## Part Info

```
firmware_size: 61440
vendor_id: 0x258a
product_id: 0x0049
bootloader_size: 4096 # necessary if not default, otherwise remove this line
page_size: 2048 # necessary if not default, otherwise remove this line
isp_usage_page: 0xff00 # necessary if not default, otherwise remove this line
isp_usage: 0x0001 # necessary if not default, otherwise remove this line
isp_index: 0 # necessary if not default, otherwise remove this line
```

## Operations Tested

- [x] Read
- [ ] Write

## Platforms Tested

- [x] linux
- [ ] macos
- [ ] windows

## Checksums

- Bootloader MD5: `2d169670eae0d36eae8188562c1f66e8`
- Stock Firmware MD5: `5a02eda2240f5532d2493c81a928c861`

## HID Dump

A dump from [usbhid-dump](https://github.com/DIGImend/usbhid-dump), [win-hid-dump](https://github.com/todbot/win-hid-dump) or [mac-hid-dump](https://github.com/todbot/mac-hid-dump)

<details>
<summary>HID Tool Output</summary>

```
# K500-B61 using usbhid-dump
...
001:008:001:DESCRIPTOR         1711664651.046945
 06 01 00 09 80 A1 01 85 01 19 81 29 83 15 00 25
 01 95 03 75 01 81 02 95 01 75 05 81 01 C0 05 0C
 09 01 A1 01 85 02 19 00 2A FF 02 15 00 26 FF 7F
 95 01 75 10 81 00 C0 06 00 FF 09 01 A1 01 85 03
 15 00 26 FF 00 09 2F 75 08 95 03 81 02 C0 05 01
 09 06 A1 01 85 04 05 07 19 04 29 70 15 00 25 01
 75 01 95 78 81 02 C0 06 00 FF 09 01 A1 01 85 05
 15 00 26 FF 00 19 01 29 02 75 08 95 05 B1 02 C0
 06 00 FF 09 01 A1 01 85 06 15 00 26 FF 00 19 01
 29 02 75 08 96 07 04 B1 02 C0 05 01 09 02 A1 01
 85 07 09 01 A1 00 05 09 15 00 25 01 19 01 29 05
 75 01 95 05 81 02 95 03 81 01 05 01 16 00 80 26
 FF 7F 09 30 09 31 75 10 95 02 81 06 15 81 25 7F
 09 38 75 08 95 01 81 06 05 0C 0A 38 02 95 01 81
 06 C0 C0

001:008:000:DESCRIPTOR         1711664651.049886
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 95 08 75 01 81 02 95 01 75 08 81 03 95 06 75 08
 15 00 26 FF 00 05 07 19 00 2A FF 00 81 00 25 01
 95 05 75 01 05 08 19 01 29 05 91 02 95 01 75 03
 91 03 C0
...
```

</details>

carlossless, 2024-03-29:
@wyrla thanks for the report!

It's currently not possible to write a full firmware image (main fw + bootloader) using this tool. (I am assuming it's complete/full image due to the filename). Try writing the image you get when using the default reading mode (bootloader portion excluded).

wyrla, 2024-03-29:
@carlossless you're right, it works as a charm now. I am going to update the report, thanks a lot!

closed 2024-03-29 by carlossless