karstx/sinowisp #61: [device-report] Manufacturer Model

Sherman729, 2024-04-04:
## Device Info

- Sinowealth Device: `SH68F90A`
- IC Label: `BYK916`
- Product Page: _https://redragonshop.com/products/irelia-k658-pro-90-full-transparent-keyboard_

## Part Info

```
firmware_size: 61440
vendor_id: 0x258a
product_id: 0x49
```

## Operations Tested

- [ ok] Read
- [ ok] Write (Linux Mint only)

## Platforms Tested

- [ ok] linux
- [ ] macos
- [ ok] windows

## Checksums

- Bootloader MD5: `3e0ebd0c440af5236d7ff8872343f85d`
- Stock Firmware MD5: `da07d2bde42817f2c8ac51fdbe49c3d2`

## HID Dump

A dump from [usbhid-dump](https://github.com/DIGImend/usbhid-dump), [win-hid-dump](https://github.com/todbot/win-hid-dump) or [mac-hid-dump](https://github.com/todbot/mac-hid-dump)

<details>
<summary>HID Tool Output</summary>

```
# Redragon K658 Irelia Pro using win-hid-dump
...
258A: 0049: BY Tech - Gaming Keyboard
PATH:\\?\hid#vid_258a&pid_0049&mi_01&col03#8&22f876a5&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
06 00 FF 09 01 A1 01 85 03 09 2F 15 00 25 FF 35 
00 45 00 65 00 55 00 75 08 95 03 81 02 C1 00 
(31 bytes)

...
```

</details>

carlossless, 2024-04-05:
Thanks for the report @Sherman729!

Could you please share the HID Dump for the keyboard when it is in its stock firmware mode (not bootloader mode)?

Sherman729, 2024-04-08:
Hi! Could this be the right one?
[image]

Sherman729, 2024-04-10:
Deleted the old one, added the new. Need the bootloader one too?

carlossless, 2024-04-13:
@Sherman729 sorry, I might have mistaken your previous dump for when the device was in bootloader mode.

Still, since windows divides these HID devices into multiple TLCs I would like to see all records, but only one of them is shared.

Can you please add all records that have the path prefix `PATH:\\?\hid#vid_258a&pid_0049` into the HID Tool Output?

Sherman729, 2024-04-13:
The HID dumper crashes a lot, but let me get it running in a clean machine
and I'll go back to it.

Sherman729, 2024-04-13:
I got it to work in a clean linux install, kb is the only device attached to the machine. Dumped with usbhid-dump, as follows:
001:010:001:DESCRIPTOR         1713041695.200599
 06 01 00 09 80 A1 01 85 01 19 81 29 83 15 00 25
 01 95 03 75 01 81 02 95 01 75 05 81 01 C0 05 0C
 09 01 A1 01 85 02 19 00 2A FF 02 15 00 26 FF 7F
 95 01 75 10 81 00 C0 06 00 FF 09 01 A1 01 85 03
 15 00 26 FF 00 09 2F 75 08 95 03 81 02 C0 05 01
 09 06 A1 01 85 04 05 07 19 04 29 70 15 00 25 01
 75 01 95 78 81 02 C0 06 00 FF 09 01 A1 01 85 05
 15 00 26 FF 00 19 01 29 02 75 08 95 05 B1 02 C0
 06 00 FF 09 01 A1 01 85 06 15 00 26 FF 00 19 01
 29 02 75 08 96 07 04 B1 02 C0 05 01 09 02 A1 01
 85 07 09 01 A1 00 05 09 15 00 25 01 19 01 29 05
 75 01 95 05 81 02 95 03 81 01 05 01 16 00 80 26
 FF 7F 09 30 09 31 75 10 95 02 81 06 15 81 25 7F
 09 38 75 08 95 01 81 06 05 0C 0A 38 02 95 01 81
 06 C0 C0 06 00 FF 09 01 A1 01 85 08 15 00 26 FF
 00 09 00 75 08 96 7D 01 B1 02 C0

001:010:000:DESCRIPTOR         1713041695.202649
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 95 08 75 01 81 02 95 01 75 08 81 03 95 06 75 08
 15 00 26 FF 00 05 07 19 00 2A FF 00 81 00 25 01
 95 05 75 01 05 08 19 01 29 05 91 02 95 01 75 03
 91 03 C0

carlossless, 2024-04-13:
Yes, thank you!

Sherman729, 2024-04-13:
Glad to hear it!

closed 2024-04-14 by carlossless