karstx/sinowisp #64: [bug] Machinike K550 B61

wyrla, 2024-04-07:
Hi, @carlossless, just to add a note about report #58, yesterday I was testing sinowealth using the ` -p machenike-k500-b61` and the `read` operation was successful but the `write` was returning a mismatch firewall error, previous I was using the flag `-p redragon-k617-fizz` and both operation succeed without problems. 
Another detail, I verified another Machenike keyboard [K500 B94](https://global.machenike.com/products/k500), and it uses the same mcu (BYK916) and also has the idVendor and idProduct as 0x258a and 0x0049, but I couldn't test `sinowealth-kb-tool` in them (keyboard isn't mine).

carlossless, 2024-04-07:
@wyrla huh, that seems unexpected...

Could you provide:
* full logs of the command that fails (with `DEBUG=1`)
* what os are you using?
* which version of sinowealth-kb-tool are you using?

wyrla, 2024-04-07:
I was using sinowealth-kb-tool v0.0.10 and my os is Ubuntu 22.04.4 LTS. I can't provide logs because I bricked my keyboard (I knew I would do it at some point).

fucksophie, 2024-04-12:
Hello, I had read the firmware and then I updated to the newest Machenike firmware, it didn't work and was very buggy, and then I wrote the old firmware with no issues. No clue what you ran into, might be related with the fact that it's bricked.

fucksophie, 2024-04-12:
> I was using sinowealth-kb-tool v0.0.10 and my os is Ubuntu 22.04.4 LTS. I can't provide logs because I bricked my keyboard (I knew I would do it at some point).

Out of curiosity, how did you brick it?

carlossless, 2024-04-13:
@wyrla

> a mismatch firewall error

I assume you meant a `Firmware Mismatch` error?

Could you please retrace your steps if you think sinowealth-kb-tool was the cause of your device being bricked?
Also, do you have a dump of the firmware you were trying to write and got this error? Send it to: contact [at] carlossless.io

@fucksophie

So you have the same device and could confirm read and write was working fine with this tool?

fucksophie, 2024-04-13:
Yes, R/W is working perfectly fine with this tool for this board.

wyrla, 2024-04-15:
> @wyrla
>
>
> I assume you meant a `Firmware Mismatch` error?
>
> Could you please retrace your steps if you think sinowealth-kb-tool was the cause of your device being bricked? Also, do you have a dump of [...]

Thats right! I don't know exactly how happened, I was playing with the firmware and it got buggy so tried to use the `write` operation to write my `.hex` generated by the `sinowealth-kb-tool` so I got the `firmware mismatch` error, so tried to use hex file I got when I tested `sinowealth-kb-tool v9` with the `redragon-k617-fizz` flag, everything worked flawlessly so I tried to reread the bootleader using the `machenike-k500-b61` (it worked without problems) then I tried to write another time and I got the same error and I decided to run the write operation one more time (here is when I made my mistake) and I receive the firmware mismatch error and the keyboard turned off and stopped being recognized. I don't think it was `sinowealth-kb-tool` the reason, but I did feel responsible because I opened the device report here. I didn't had how to test one more to exclude the _me_ variable from the equation so I should at least report again to ensure everything was all right. 

> Yes, R/W is working perfectly fine with this tool for this board.

With @fucksophie response I believe that we can close this topic because I feared that the error was something generalized (caused by some missing information in the device report) and not caused by me, sorry if it seemed I was blaming you, @carlossless. It's not like that, your work here is phenomenal.

carlossless, 2024-04-16:
@wyrla 

> sorry if it seemed I was blaming you, @carlossless.

No worries, I didn't take it that way! ❤️

These ISP bootloaders and how they work are pretty brittle, so it's not hard to brick your device with firmware that was modified in all of the right (or rather wrong) places :). I have done this many times in my initial testing.

Still, there are scenarios where I would be very much surprised if it happened to lead to a bricked device. Aborting a write operation mid-way, being one of them.

I am going to close this issue for now, but feel free to comment and reopen it if you happen to reproduce the steps that bricked your device. Even if it's not an issue with this tool, it might help the next person who runs into something similar!

closed 2024-04-16 by carlossless

fucksophie, 2024-04-16:
this is most likely not the correct place, but where did you buy your sinolink from? they seem to cost around 100€, and i bought my whole keyboard for 60€ + I'm kind of a student so buying a thing I'm going to use for working on a single keyboard seems kind of excessive.. is there any alternative (with documentation)?

carlossless, 2024-04-16:
@fucksophie I got mine through [taobao](https://world.taobao.com/) + [superbuy](https://www.superbuy.com/). [the listing I purchased from is still up](https://www.superbuy.com/en/page/buy/?url=https%3A%2F%2Fitem.taobao.com%2Fitem.htm%3Fid%3D558627137063%26ali_trackid%3D2%3Amm_117358474_33384934_118778578). All in all I think I spent a little bit over 60eur.

> I'm kind of a student so buying a thing I'm going to use for working on a single keyboard seems kind of excessive.. is there any alternative (with documentation)?

Unfortunately not at the moment :/

Though, I know @swiftgeek has been working on something related. Possibly an open-hardware alternative to the official sinolink.
There's also [gashtaan/sinowealth-8051-bl-updater](https://github.com/gashtaan/sinowealth-8051-bl-updater) which includes writing subroutines through JTAG built-in and could maybe be extended to a full programming tool like the sinolink.