karstx/sinowisp #65: [device-report] Xinmeng M71 v2 (same with Yunzii AL71)

rfcclub, 2024-04-08:
## Device Info

- Sinowealth Device: SH6890A
- IC Label: Sinowealth SH6890AS 
- PCB label: BYK926
- Product Page: https://www.yunzii.com/en-ca/products/yunzii-al71-mechanical-keyboard

## Part Info

```
firmware_size: 61440
vendor_id: 0x258a
product_id: 0x010c
bootloader_size: 4096 # necessary if not default, otherwise remove this line
page_size: 2048 # necessary if not default, otherwise remove this line
isp_usage_page: 0xff00 # necessary if not default, otherwise remove this line
isp_usage: 0x0001 # necessary if not default, otherwise remove this line
isp_index: 0 # necessary if not default, otherwise remove this line
```

## Operations Tested

- [ ] Read
- [ ] Write

## Platforms Tested

- [ ] linux
- [ ] macos
- [ x] windows

## Checksums

- Bootloader MD5: Unknown (eg `beefcafebeefcafebeefcafebeefcafe`)
- Stock Firmware MD5: Unknown (eg `deadbeefdeadbeefdeadbeefdeadbeef`)

## HID Dump

A dump from [usbhid-dump](https://github.com/DIGImend/usbhid-dump), [win-hid-dump](https://github.com/todbot/win-hid-dump) or [mac-hid-dump](https://github.com/todbot/mac-hid-dump)

<details>
<summary>HID Tool Output</summary>

```
# Xinmeng M71 using win-hid-dump
...
258A:010C: BY Tech - Gaming Keyboard
PATH:\\?\hid#vid_258a&pid_010c&mi_01&col06#b&4d15b8b&0&0005#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  06  00  FF  09  01  A1  01  85  06  15  00  25  01  35  00  45
  01  65  00  55  00  75  01  95  38  81  03  96  38  10  B1  03
  C1  00
  (34 bytes)
...
258A:010C: BY Tech - Gaming Keyboard
PATH:\\?\hid#vid_258a&pid_010c&mi_01&col07#b&4d15b8b&0&0006#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  05  01  09  02  A1  01  85  07  09  01  A1  00  05  09  19  01
  29  05  15  00  25  01  35  00  45  01  65  00  55  00  75  01
  95  05  81  02  95  03  81  03  05  01  09  30  26  FF  7F  45
  00  75  10  95  01  81  06  09  31  81  06  09  38  25  7F  75
  08  81  06  05  0C  0A  38  02  81  06  C1  00  C1  00
  (78 bytes)
258A:010C: BY Tech - Gaming Keyboard
PATH:\\?\hid#vid_258a&pid_010c&mi_01&col05#b&4d15b8b&0&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  06  00  FF  09  01  A1  01  85  05  15  00  25  01  35  00  45
  01  65  00  55  00  75  01  95  28  B1  03  C1  00
  (29 bytes)
...
258A:010C: BY Tech - Gaming Keyboard
PATH:\\?\hid#vid_258a&pid_010c&mi_01&col01#b&4d15b8b&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  05  01  09  80  A1  01  85  01  19  81  29  83  15  00  25  01
  35  00  45  01  65  00  55  00  75  01  95  03  81  02  95  05
  81  03  C1  00
  (36 bytes)
258A:010C: BY Tech - Gaming Keyboard
PATH:\\?\hid#vid_258a&pid_010c&mi_01&col03#b&4d15b8b&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  06  00  FF  09  01  A1  01  85  03  09  2F  15  00  25  FF  35
  00  45  00  65  00  55  00  75  08  95  03  81  02  C1  00
  (31 bytes)
258A:010C: BY Tech - Gaming Keyboard
PATH:\\?\hid#vid_258a&pid_010c&mi_00#b&e5ec0b3&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
DESCRIPTOR:
  05  01  09  06  A1  01  05  07  19  E0  29  E7  15  00  25  01
  35  00  45  01  65  00  55  00  75  01  95  08  81  02  95  38
  81  03  05  08  19  01  29  05  95  05  91  02  95  03  91  03
  C1  00
  (50 bytes)
258A:010C: BY Tech - Gaming Keyboard
PATH:\\?\hid#vid_258a&pid_010c&mi_01&col02#b&4d15b8b&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  05  0C  09  01  A1  01  85  02  15  00  25  01  35  00  45  01
  65  00  55  00  75  01  95  10  81  03  C1  00
  (28 bytes)
258A:010C: BY Tech - Gaming Keyboard
PATH:\\?\hid#vid_258a&pid_010c&mi_01&col04#b&4d15b8b&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
DESCRIPTOR:
  05  01  09  06  A1  01  85  04  05  07  19  04  29  70  15  00
  25  01  35  00  45  01  65  00  55  00  75  01  95  6D  81  02
  95  0B  81  03  C1  00
  (38 bytes)
```

</details>

rfcclub, 2024-04-08:
I've run sinowealth-kb-tool 0.0.10 and it returned nothing. Will try to run it on a Linux distro.
```
sinowealth-kb-tool.exe read --vendor_id 0x258a --product_id 0x010c --firmware_size 61440 xm71.hex
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x010c
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x010c
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 3/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x010c
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 4/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x010c
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 5/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x010c
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 6/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x010c
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 7/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x010c
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 8/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x010c
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 9/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x010c
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 10/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x010c
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
ERROR [sinowealth_kb_tool] Device not found
```

carlossless, 2024-04-13:
@rfcclub based on the HID dump, try using `--isp_index 1`, it should work:
```sh
sinowealth-kb-tool.exe read --vendor_id 0x258a --product_id 0x010c --firmware_size 61440 --isp_index 1 xm71.hex
```

rfcclub, 2024-04-14:
It worked, but I performed the read through an USB hub so I must plug keyboard directly to PC to make it works again. Read is OK, I am not confidence to perform a write yet since I don't have Sinolink to backup it first.

carlossless, 2024-04-14:
@rfcclub could you please read the bootloader portion of your device and update the device report template with it? (use the `-b` flag)

Also, based on the title, do you have two different devices? Do they have the exact same firmware? Eitherway, could you please open up another device-report issue for the Yunzii AL71, if you're able to test against it?

> I am not confidence to perform a write yet since I don't have Sinolink to backup it first.

You can get a full dump of your device flash with the `--full` option, but yes, incase something goes wrong, you would need other means to restore it (sinolink).

rfcclub, 2024-04-18:
@carlossless  here is full dump with log
[```]
[xm71.zip](https://github.com/carlossless/sinowealth-kb-tool/files/15023799/xm71.zip)

sinowealth-kb-tool.exe read --vendor_id 0x258a --product_id 0x010c --full --firmware_size 61440 --isp_index 1 xm71.hex
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x010c
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Waiting for ISP device...
INFO  [sinowealth_kb_tool::isp] Connected!
INFO  [sinowealth_kb_tool::isp] Enabling firmware...
INFO  [sinowealth_kb_tool::isp] Reading...
INFO  [sinowealth_kb_tool::isp] Rebooting...
INFO  [sinowealth_kb_tool] MD5: c6b1be8743ec87eb0c3aa630939c6e41
```
Actually I read from community AL71 and M71 are the same. Also I see they share the same software (AL71 also has M71 directory in their software). I can just try on M71 only.
![image](https://github.com/carlossless/sinowealth-kb-tool/assets/1235656/d963deea-d174-408b-b15d-8f9342570093)

rfcclub, 2024-04-18:
Log for reading with boot_loader flag
```
sinowealth-kb-tool.exe read --vendor_id 0x258a --product_id 0x010c --firmware_size 61440 --bootloader_size 4096 --isp_index 1 xm71.hex
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x010c
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Waiting for ISP device...
INFO  [sinowealth_kb_tool::isp] Connected!
INFO  [sinowealth_kb_tool::isp] Enabling firmware...
INFO  [sinowealth_kb_tool::isp] Reading...
INFO  [sinowealth_kb_tool::isp] Rebooting...
INFO  [sinowealth_kb_tool] MD5: 189f254a5715fba9ccd610890f5b90fc
```

[xm71_bootloader.zip](https://github.com/carlossless/sinowealth-kb-tool/files/15023855/xm71_bootloader.zip)
Hope it helps to determine device information.

carlossless, 2024-04-27:
@rfcclub thanks! the flag to read the bootloader is actually `-b` but I got the bootloader portion from your full dump and can confirm that its md5sum is `2d169670eae0d36eae8188562c1f66e8`!

carlossless, 2024-04-27:
@rfcclub interestingly I noticed that the md5sum for your first read (`a123abac3a075fd4745cb6fd31a5324d`) and second read (`189f254a5715fba9ccd610890f5b90fc`) is different. 
The only actual difference is:
[image]

🤔