karstx/sinowisp #66: [device-report] E-Yooso Z-11 with yellow LEDs

BenoitPerron, 2024-04-10:
## Device Info

- Sinowealth Device: `SH68F90?`
- IC Label: _example `BYK910`
- Product Page: https://www.aliexpress.com/item/1005005775806125.html

## Part Info

```
firmware_size: 168972
vendor_id: 0x258a
product_id: 0x002a
bootloader_size: 11276

```

## Operations Tested

- [x] Read
- [ ] Write

## Platforms Tested

- [x] linux
- [ ] macos
- [ ] windows

## Checksums

- Bootloader MD5: `3e0ebd0c440af5236d7ff8872343f85d`
- Stock Firmware MD5: `03205dbb8dc26354fe04786d0d7f7625`

## HID Dump

A dump from [usbhid-dump](https://github.com/DIGImend/usbhid-dump), [win-hid-dump](https://github.com/todbot/win-hid-dump) or [mac-hid-dump](https://github.com/todbot/mac-hid-dump)

<details>
<summary>HID Tool Output</summary>

```
sudo usbhid-dump --entity=all --model=258a
003:021:001:DESCRIPTOR         1712708430.881304
 05 01 09 80 A1 01 85 01 19 81 29 83 15 00 25 01
 75 01 95 03 81 02 95 05 81 01 C0 05 0C 09 01 A1
 01 85 02 19 00 2A 3C 02 15 00 26 3C 02 95 01 75
 10 81 00 C0 06 00 FF 09 01 A1 01 85 05 15 00 26
 FF 00 19 01 29 02 75 08 95 05 B1 02 C0 05 01 09
 06 A1 01 85 06 05 07 19 04 29 70 15 00 25 01 75
 01 95 78 81 02 C0 06 00 FF 09 01 A1 01 85 09 15
 00 26 FF 00 09 00 75 08 96 F8 01 B1 02 C0 06 00
 FF 09 01 A1 01 85 0A 15 00 26 FF 00 09 00 75 08
 95 29 B1 02 C0 06 00 FF 09 01 A1 01 85 0B 15 00
 26 FF 00 09 00 75 08 95 7E B1 02 C0 05 01 09 02
 A1 01 85 0D 09 01 A1 00 05 09 15 00 25 01 19 01
 29 05 75 01 95 05 81 02 95 03 81 01 05 01 16 00
 80 26 FF 7F 09 30 09 31 75 10 95 02 81 06 15 81
 25 7F 09 38 75 08 95 01 81 06 05 0C 0A 38 02 95
 01 81 06 C0 C0 06 00 FF 09 01 A1 01 85 0C 15 00
 26 FF 00 09 00 75 08 96 80 07 B1 02 C0

003:021:000:DESCRIPTOR         1712708430.882968
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 95 08 75 01 81 02 95 01 75 08 81 03 95 06 75 08
 15 00 26 FF 00 05 07 19 00 2A FF 00 81 00 25 01
 95 05 75 01 05 08 19 01 29 05 91 02 95 01 75 03
 91 03 C0
```

</details>

## Other Outputs

<details>
<summary>Dmesg Output</summary>

```
[339005.632758] usb 3-3: new full-speed USB device number 22 using xhci_hcd
[339005.775552] usb 3-3: New USB device found, idVendor=258a, idProduct=002a, bcdDevice=12.10
[339005.775566] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[339005.775571] usb 3-3: Product: Gaming KB
[339005.775574] usb 3-3: Manufacturer: SINO WEALTH
[339005.778909] input: SINO WEALTH Gaming KB  as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.0/0003:258A:002A.002C/input/input103
[339005.833222] hid-generic 0003:258A:002A.002C: input,hidraw0: USB HID v1.11 Keyboard [SINO WEALTH Gaming KB ] on usb-0000:00:14.0-3/input0
[339005.838636] input: SINO WEALTH Gaming KB  System Control as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.1/0003:258A:002A.002D/input/input104
[339005.893098] input: SINO WEALTH Gaming KB  Consumer Control as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.1/0003:258A:002A.002D/input/input105
[339005.893395] input: SINO WEALTH Gaming KB  Keyboard as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.1/0003:258A:002A.002D/input/input106
[339005.893683] input: SINO WEALTH Gaming KB  Mouse as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.1/0003:258A:002A.002D/input/input107
[339005.894348] hid-generic 0003:258A:002A.002D: input,hiddev96,hidraw1: USB HID v1.11 Keyboard [SINO WEALTH Gaming KB ] on usb-0000:00:14.0-3/input1
```

</details>

<details>
<summary>sinowealth-kb-tool read -p xinmeng-xm-rf68 z11.hex Output</summary>

```
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x002a
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x002a
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Connected!
INFO  [sinowealth_kb_tool::isp] Enabling firmware...
INFO  [sinowealth_kb_tool::isp] Reading...
INFO  [sinowealth_kb_tool::isp] Rebooting...
INFO  [sinowealth_kb_tool] MD5: 03205dbb8dc26354fe04786d0d7f7625
```

</details>

<details>
<summary>sinowealth-kb-tool read -p xinmeng-xm-rf68 -b z11-boot.hex Output</summary>

```
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x002a
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x002a
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Connected!
INFO  [sinowealth_kb_tool::isp] Enabling firmware...
INFO  [sinowealth_kb_tool::isp] Reading...
INFO  [sinowealth_kb_tool::isp] Rebooting...
INFO  [sinowealth_kb_tool] MD5: 3e0ebd0c440af5236d7ff8872343f85d
```

</details>

carlossless, 2024-04-13:
@BenoitPerron thanks for the report!

> bootloader_size: 11276

Is this part a mistake? It seems like your device has a bootloader that was already encountered on other devices (including my own), but all of their sizes were 4096 bytes.

BenoitPerron, 2024-04-15:
Sorry, my bad! the boot loaded is indeed listed.
Stupid question: How do you get the dump size in bytes? I naively reported the size of the hex file!

carlossless, 2024-04-16:
@BenoitPerron

The size of your payloads is already predetermined by the configuration that's either built-in or given through arguments to the tool, so the resulting payloads should never be different. Still if you'd like to confirm, you can either:

Check the last address of the ihex that this tool outputs. Here's an example:

```ihex
:10EFF000000000000000000000000002007100009E
:00000001FF
```

0xEFF0 (start address) + 0x10 (16 in dec, size of content per line) = 0xF000 = 61440

OR

You can convert the hex into a binary:

```sh
objcopy --input-target=ihex --output-target=binary foobar.hex foobar.bin
```

The resulting filesize of the binary file will be equivalent to the size of the payload

closed 2024-04-16 by carlossless