karstx/sinowisp #67: [device-report] Magebee K61

michal4132, 2024-04-10:
Doesn't enumerate as ISP in dmesg. From HID descriptor it should work with `usage_page == 0xff00 && usage == 0x0001`.

## Device Info

- Sinowealth Device: ?
- IC Label:  `BYK903`
- PCB Label: `BYK901`
- Product Page: https://pl.aliexpress.com/item/1005004869547787.html

## Part Info

```
firmware_size: ?
vendor_id: 0x258a
product_id: 0x013b
bootloader_size: ? # necessary if not default, otherwise remove this line
page_size: ? # necessary if not default, otherwise remove this line
isp_usage_page: ? # necessary if not default, otherwise remove this line
isp_usage: ? # necessary if not default, otherwise remove this line
isp_index: ? # necessary if not default, otherwise remove this line
```

## Operations Tested

- [ ] Read
- [ ] Write

## Platforms Tested

- [X] linux
- [ ] macos
- [ ] windows

## Checksums

- Bootloader MD5: `beefcafebeefcafebeefcafebeefcafe`
- Stock Firmware MD5: `deadbeefdeadbeefdeadbeefdeadbeef`

## HID Dump

A dump from [usbhid-dump](https://github.com/DIGImend/usbhid-dump), [win-hid-dump](https://github.com/todbot/win-hid-dump) or [mac-hid-dump](https://github.com/todbot/mac-hid-dump)

<details>
<summary>HID Tool Output</summary>

```
001:099:001:DESCRIPTOR         1712737273.934683
 05 01 09 80 A1 01 85 01 19 81 29 83 15 00 25 01
 75 01 95 03 81 02 95 05 81 01 C0 05 0C 09 01 A1
 01 85 02 19 00 2A 3C 02 15 00 26 3C 02 95 01 75
 10 81 00 C0 06 00 FF 09 01 A1 01 85 05 15 00 26
 FF 00 19 01 29 02 75 08 95 05 B1 02 C0 05 01 09
 06 A1 01 85 06 05 07 19 04 29 70 15 00 25 01 75
 01 95 78 81 02 C0 06 00 FF 09 01 A1 01 85 09 15
 00 26 FF 00 09 00 75 08 96 F8 01 B1 02 C0 06 00
 FF 09 01 A1 01 85 0A 15 00 26 FF 00 09 00 75 08
 95 29 B1 02 C0 06 00 FF 09 01 A1 01 85 0B 15 00
 26 FF 00 09 00 75 08 95 7E B1 02 C0 05 01 09 02
 A1 01 85 0D 09 01 A1 00 05 09 15 00 25 01 19 01
 29 05 75 01 95 05 81 02 95 03 81 01 05 01 16 00
 80 26 FF 7F 09 30 09 31 75 10 95 02 81 06 15 81
 25 7F 09 38 75 08 95 01 81 06 05 0C 0A 38 02 95
 01 81 06 C0 C0 06 00 FF 09 01 A1 01 85 0C 15 00
 26 FF 00 09 00 75 08 96 80 07 B1 02 C0

001:099:000:DESCRIPTOR         1712737273.936511
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 95 08 75 01 81 02 95 01 75 08 81 03 95 06 75 08
 15 00 26 FF 00 05 07 19 00 2A FF 00 81 00 25 01
 95 05 75 01 05 08 19 01 29 05 91 02 95 01 75 03
 91 03 C0

```

</details>

carlossless, 2024-04-13:
@michal4132 interesting... What's the output of `sinowealth-kb-tool -l` when you have this device plugged in?

carlossless, 2024-04-16:
@michal4132 alright, nothing unusual there.

Could you please share the logs you get when running the tool against your device with `DEBUG=1`?

michal4132, 2024-04-17:
Sure, here are logs:
`sudo DEBUG=1 sinowealth-kb-tool read --vendor_id 0x258a --product_id 0x013b --isp_usage_page 0xff00 --isp_usage 0x0001 --firmware_size 61440 out.hex`
```
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x013b
DEBUG [sinowealth_kb_tool::isp] Found Device: "1-3.1:1.1"
DEBUG [sinowealth_kb_tool::isp] Opening: "1-3.1:1.1"
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x013b
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 3/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x013b
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 4/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x013b
DEBUG [sinowealth_kb_tool::isp] Found Device: "1-3.1:1.1"
DEBUG [sinowealth_kb_tool::isp] Opening: "1-3.1:1.1"
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 5/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x013b
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 6/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x013b
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 7/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x013b
DEBUG [sinowealth_kb_tool::isp] Found Device: "1-3.1:1.1"
DEBUG [sinowealth_kb_tool::isp] Opening: "1-3.1:1.1"
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 8/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x013b
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 9/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x013b
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 10/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x013b
DEBUG [sinowealth_kb_tool::isp] Found Device: "1-3.1:1.1"
DEBUG [sinowealth_kb_tool::isp] Opening: "1-3.1:1.1"
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
ERROR [sinowealth_kb_tool] Device not found
```

carlossless, 2024-04-28:
@michal4132 I added more logging in v0.0.11 which should hopefully give us more information on what error is actually occurring here. Please try the same command with that version.

michal4132, 2024-04-30:
I've tried v0.0.11 but hidapi errors are not implemented. Is it possible that I'm using old hidapi?
```
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x013b
DEBUG [sinowealth_kb_tool::isp] Found Device: "1-5:1.1"
DEBUG [sinowealth_kb_tool::isp] Opening: "1-5:1.1"
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
DEBUG [sinowealth_kb_tool::isp] Error: hidapi error: hid_error is not implemented yet
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x013b
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 3/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x013b
DEBUG [sinowealth_kb_tool::isp] Found Device: "1-5:1.1"
DEBUG [sinowealth_kb_tool::isp] Opening: "1-5:1.1"
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
DEBUG [sinowealth_kb_tool::isp] Error: hidapi error: hid_error is not implemented yet
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 4/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x013b
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 5/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x013b
DEBUG [sinowealth_kb_tool::isp] Found Device: "1-5:1.1"
DEBUG [sinowealth_kb_tool::isp] Opening: "1-5:1.1"
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
DEBUG [sinowealth_kb_tool::isp] Error: hidapi error: hid_error is not implemented yet
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 6/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x013b
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 7/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x013b
DEBUG [sinowealth_kb_tool::isp] Found Device: "1-5:1.1"
DEBUG [sinowealth_kb_tool::isp] Opening: "1-5:1.1"
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
DEBUG [sinowealth_kb_tool::isp] Error: hidapi error: hid_error is not implemented yet
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 8/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x013b
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 9/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x013b
DEBUG [sinowealth_kb_tool::isp] Found Device: "1-5:1.1"
DEBUG [sinowealth_kb_tool::isp] Opening: "1-5:1.1"
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
DEBUG [sinowealth_kb_tool::isp] Error: hidapi error: hid_error is not implemented yet
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 10/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x013b
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
ERROR [sinowealth_kb_tool] Device not found
```