karstx/sinowisp #68: [question] What do I need to watch out when flashing?

ViktorTrojan, 2024-04-13:
I was wondering if someone could give me insights on what I should be watching out when flashing. I have a RK84 RGB ISO Return Keyboard and dumped both the isp and bootloader. I would like to try and flash smk onto it but worry something could go wrong. Does smk also change the bootloader? and if it doesnt and I still mess up is it then possible to enter bootloader and flash back the original firmware with this tool? or do I need to physically open the device and short the reset pins or something like that?

carlossless, 2024-04-13:
@ViktorTrojan 

This tool won't overwrite the bootloader. Due to the way these ISP bootloaders work, if flashing main firmware fails midway, the keyboard will continue to operate in bootloader-only mode, until a new firmware is successfully written to the device.

That said...

**If you successfully flash "broken" firmware that doesn't have a working subroutine to get you back into ISP mode, you won't be able to overwrite it over USB again.**

If you get to that state you will only be able to restore your original firmware with a dedicated programmer tool like the sinolink and the prowriter software.

Also, I don't recommend using SMK at this point unless you're interested in experimentation & development and understand that it might brick your device.

ViktorTrojan, 2024-04-29:
> @ViktorTrojan
>
> This tool won't overwrite the bootloader. Due to the way these ISP bootloaders work, if flashing main firmware fails midway, the keyboard will continue to operate in bootloader-only [...]

is it possible to do it with some arduino? would like to know more about this topic but cant find much online

carlossless, 2024-05-12:
> is it possible to do it with some arduino? would like to know more about this topic but cant find much online

Yes, projects like https://github.com/gashtaan/sinowealth-8051-dumper & https://github.com/gashtaan/sinowealth-8051-bl-updater are basically that but need a little more work to allow you to write full dumps to the devices (among possibly other things).

closed 2024-06-24 by carlossless