karstx/sinowisp #71: [bug] Custom vendor read mode - Device Not Found

Lisoveliy, 2024-04-20:
Hello. I tried to read firmware from Redragon CASTOR PRO (K631RGB-PRO-BRW) and I got this error:

[image]

As you can see, my device was displayed in the "list" command, but when I tried to read it, it failed. I tried to use the "firmware-size" argument instead of the "part" argument, but it also failed. Can you give me advice on why it completed unsuccessfully and what the "firmware-size" argument means? Where can I get this value? Thanks.

Lisoveliy, 2024-04-20:
By the way, I think K641 and K631 are very similar, because in technical information (as I checked) about these keyboards, they have only one difference between them - the material of the casing.

carlossless, 2024-04-20:
@Lisoveliy can you please open up a new [device-report](https://github.com/carlossless/sinowealth-kb-tool/issues/new?assignees=&labels=device-report&projects=&template=device-report.md&title=%5Bdevice-report%5D+Manufacturer+Model) issue and add all known details for your device, particularly the "HID Dump" section? I assume the isp_index will be different from k641 (isp_index=0 instead of isp_index=1).

closed 2024-04-20 by carlossless