karstx/sinowisp #72: [device-report] maybe Dareu EK681

lincoo2436, 2024-04-21:
## Device Info

- Sinowealth Device:  `SH68F90A`_
- IC Label: Sinowealth `SH68F90A`
- Product Page: _example https://nuphy.com/products/air60_

## Part Info

```
firmware_size: 61440
vendor_id: 0x258a
product_id: 0x0049

```

## Operations Tested

- [X] Read
- [X] Write

## Platforms Tested

- [X ] linux
- [ ] macos
- [ ] windows

<details>
<summary>Need the other same vid and pid keyboard hex</summary>

```
I have a keyboard with some special reason that use the chip with ek861's chip, but the keyboard model is ek871. The keyboard can be used well except the left key replace with fn key. So I want to flash the same vid and pid keyboard's hex files to make the key recovery normal. 
Does anyone can help me? Please send me the hex file  to my email lincoo2436@gmail.com. Thx a lot!
I think the most fit keyboard is reddragon k641, the rk71 is the same layout,but the pid is not the same. I dont know may it will make bricked . Or someone can help me to compile a 71key layout with smk? When I try to make a smk. I meet the error.
![screenshot](https://github.com/carlossless/sinowealth-kb-tool/assets/26353584/f111f8d9-57fa-4477-bbfb-bac610512334) 
I cant deal with it.
I also have the ek861's hex.If someone need it , can call me.
 
```

</details>

carlossless, 2024-04-27:
@lincoo2436 can you elaborate on how you know that the chip comes from an ek861 keyboard? I'm trying to understand if the hardware is different between ek861 and ek871 or is just a firmware difference?

carlossless, 2024-04-27:
@lincoo2436 also would you mind reading the bootloader (with the `-b` argument) and sharing the MD5 value with the report?

lincoo2436, 2024-05-07:
@carlossless  when I found the fn key instead of the left key. I thought that it may be a 61 keyboard. I first tried the ek861 fn hotkey . And the function was matched well. Especially when I use Fn+O to switch the Alt,APP, Ctrl,shift keys to act as arrow keys, it fits perfectly.Also the same as rgb controling. So I guess it’s the ek861’s main control chip.
Here is the bootloader hex.
[bootloader.txt](https://github.com/carlossless/sinowealth-kb-tool/files/15234136/bootloader.txt)

P.S I also have the ek871 pcb. the ek871 white light (I dont know the rgb version) was uesd the sonix chip ,named HFD48KP500. Maybe rgb version is used the sinowealth chip.

carlossless, 2024-05-12:
@lincoo2436 thanks for the bootloader dump, it's another instance of `2d169670eae0d36eae8188562c1f66e8`.

I still don't exactly understand what device you have in your possession (and what this device report is about). As far as I get it:
You have a Dareu EK871 keyboard that has key issues (left-directional-key acts as the fn-key) and you suspect that it uses firmware meant for the EK861 keyboard, becuase this EK871 has the same chip as EK861, while others variants of EK871 (which you also own) come with a sonix arm mcu instead of a sinowealth 8051 one.

Let me know if all of that is correct!

lincoo2436, 2024-05-13:
“I apologize for not being clear earlier. The situation is this: I purchased a hot-swappable PCB, and I’m not sure if the chip has been replaced or if it was soldered incorrectly from the factory. Anyway, it seems that an ek861 chip is being used, but the PCB is of the 71 key configuration. To be precise, the chip is an ek861, but it has been mistakenly used on a PCB meant for an ek871.”
I guess the dareu's RGB PCB used the sino wealth's chip. and the ek861 and ek871 maybe has the same chip model.

lincoo2436, 2024-05-13:
The Dareu has the white light version ek871. The white light version was used the sonix arm mcu. RGB version may be used the sinowealth 8051.

carlossless, 2024-05-13:
Alright, thanks for clarifying, I think I understand fully now.

I don't think the different MCU on your board was due to a mistake. I guess the manufacturer made two different revisions of the same ek871 keyboard - one with sonix and one with sinowealth parts, but somehow ended up flashing firmware for a different board (maybe ek861).

Would you care to post pictures of the board, by the way?

Happy to keep the issue open, so that it would be more visible for someone who owns the same keyboard and might be willing to share their dump!

lincoo2436, 2024-05-13:
Sure. the board of right side in picture
[image]
 is rgb version which is used with sino wealth mcu. the board of middle is sonix mcu. The three board are the ek
871.

lincoo2436, 2024-05-13:
Add a rgb version detail picture.
[image]
[image]