karstx/sinowisp #77: [device-report] E-yooso Z11

Nono9k, 2024-04-27:
## Device Info

- Sinowealth Device: ?
- IC Label: BYK901
- Product Page: _example https://nuphy.com/products/air60_

## Part Info

```
firmware_size: 61440
vendor_id: 0x258a
product_id: 0x002a

## Operations Tested

- [X ] Read
- [ ] Write

## Platforms Tested

- [ ] linux
- [ ] macos
- [X ] windows

## Checksums

- Bootloader MD5: `4d6bedef3ca9f2885fb5f8c7385075df`
- Stock Firmware MD5: `693cd148ec436131b6638fa42fc171d6`

## HID Dump

A dump from [usbhid-dump](https://github.com/DIGImend/usbhid-dump), [win-hid-dump](https://github.com/todbot/win-hid-dump) or [mac-hid-dump](https://github.com/todbot/mac-hid-dump)

<details>
<summary>HID Tool Output</summary>

```
# WINHIDDUMP E-yooso Z11
[image]

...
WinHIDdump:
258A:002A: SINO WEALTH - Gaming KB
PATH:\\?\hid#vid_258a&pid_002a&mi_00#8&1da6645a&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
DESCRIPTOR:
  05  01  09  06  A1  01  05  07  19  E0  29  E7  15  00  25  01
  35  00  45  01  65  00  55  00  75  01  95  08  81  02  95  38
  81  03  05  08  19  01  29  05  95  05  91  02  95  03  91  03
  C1  00
  (50 bytes)
258A:002A: SINO WEALTH - Gaming KB
PATH:\\?\hid#vid_258a&pid_002a&mi_01&col04#8&5e2d815&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
DESCRIPTOR:
  05  01  09  06  A1  01  85  06  05  07  19  04  29  70  15  00
  25  01  35  00  45  01  65  00  55  00  75  01  95  6D  81  02
  95  0B  81  03  C1  00
  (38 bytes)
258A:002A: SINO WEALTH - Gaming KB
PATH:\\?\hid#vid_258a&pid_002a&mi_01&col06#8&5e2d815&0&0005#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  06  00  FF  09  01  A1  01  85  0A  09  00  15  00  25  FF  35
  00  45  00  65  00  55  00  75  08  95  29  B1  02  C1  00
  (31 bytes)
...
```

</details>

Sorry if I messed anything up I don't use these types of tools a lot lol, nor github a lot

carlossless, 2024-04-27:
Thanks for the report and the photos!

I'm surprised that the MD5 of the bootloader is different (also the firmware, but that's much more likely). Did you get those values from this tool output? Would you mind sharing a `--full` dump?

Nono9k, 2024-04-27:
## Result of Full
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x002a
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Waiting for ISP device...
INFO  [sinowealth_kb_tool::isp] Connected!
INFO  [sinowealth_kb_tool::isp] Enabling firmware...
INFO  [sinowealth_kb_tool::isp] Reading...
INFO  [sinowealth_kb_tool::isp] Rebooting...
INFO  [sinowealth_kb_tool] MD5: 4d6bedef3ca9f2885fb5f8c7385075df

<details>
<summary>Hex Filet</summary>

...
:10000000024F35026396758F0122220266CE0200EE
:100010003422220266D422220000000266DA43A8BB
:100020000175A901E4F5BAF5B6F5B4F5B875B501F1
:1000300075B9012243970843970222025F4A0000E4
:10004000000000025784900BC4ECF0A3EDF0A37401
:1000500002F0EFB4150040030203E6900062F828B6
:10006000287302013F02016702018F0201B70201FA
:10007000DF02020702022F02025702027F0202A7DA
:100080000202CF0202F702031F0203470201170216
:1000900000EF0200C70200A102036F0203970203F0
:1000A000BF900BC4E07002A3E0701553EAFE53E268
:1000B000FE900BC6E0FF7E001262DD5390FE8003CF
:1000C00043900143E20122900BC4E07002A3E07070
:1000D0001753EAFD53E2FD900BC6E0FF7E0012626B
:1000E000DD5390FD43E2022243EA0253E2FD2290F7
:1000F0000BC4E07002A3E0701753EAFB53E2FB90DD
:100100000BC6E0FF7E001262DD5390FB43E2042247
:1001100043EA0453E2FB22900BC4E07002A3E070B8
:100120001753EAF753E2F7900BC6E0FF7E00126226
:10013000DD5390F743E2082243EA0853E2F72290A6
:100140000BC4E07002A3E0701753EAEF53E2EF90A4
:100150000BC6E0FF7E001262DD5390EF43E21022F7
:1001600043EA1053E2EF22900BC4E07002A3E07068
:100170001753EADF53E2DF900BC6E0FF7E00126206
:10018000DD5390DF43E2202243EA2053E2DF229056
:100190000BC4E07002A3E0701753EBFE53E3FE9034
:1001A0000BC6E0FF7E001262DD5398FE43E301229E
:1001B00043EB0153E3FE22900BC4E07002A3E07016
:1001C0001753EBFD53E3FD900BC6E0FF7E00126278
:1001D000DD5398FD43E3022243EB0253E3FD2290FB
:1001E0000BC4E07002A3E0701753EBFB53E3FB90EA
:1001F0000BC6E0FF7E001262DD5398FB43E304224E
:1002000043EB0453E3FB22900BC4E07002A3E070C5
:100210001753EBF753E3F7900BC6E0FF7E00126233
:10022000DD5398F743E3082243EB0853E3F72290AA
:100230000BC4E07002A3E0701753EBEF53E3EF90B1
:100240000BC6E0FF7E001262DD5398EF43E31022FD
:1002500043EB1053E3EF22900BC4E07002A3E07075
:100260001753EBDF53E3DF900BC6E0FF7E00126213
:10027000DD5398DF43E3202243EB2053E3DF22905A
:100280000BC4E07002A3E0701753ECFE53E4FE9041
:100290000BC6E0FF7E001262DD53A0FE43E40122A4
:1002A00043EC0153E4FE22900BC4E07002A3E07023
:1002B0001753ECFD53E4FD900BC6E0FF7E00126285
:1002C000DD53A0FD43E4022243EC0253E4FD2290FF
:1002D0000BC4E07002A3E0701753ECFB53E4FB90F7
:1002E0000BC6E0FF7E001262DD53A0FB43E4042254
:1002F00043EC0453E4FB22900BC4E07002A3E070D3
:100300001753ECF753E4F7900BC6E0FF7E00126240
:10031000DD53A0F743E4082243EC0853E4F72290AE
:100320000BC4E07002A3E0701753ECEF53E4EF90BE
:100330000BC6E0FF7E001262DD53A0EF43E4102203
:1003400043EC1053E4EF22900BC4E07002A3E07082
:100350001753ECDF53E4DF900BC6E0FF7E00126220
:10036000DD53A0DF43E4202243EC2053E4DF22905E
:100370000BC4E07002A3E0701753EEFE53E6FE904C
[... 4018 lines cut ...]
:10FEA0000000000000000000000000000000000052
:10FEB0000000000000000000000000000000000042
:10FEC0000000000000000000000000000000000032
:10FED0000000000000000000000000000000000022
:10FEE0000000000000000000000000000000000012
:10FEF0000000000000000000000000000000000002
:10FF0000645A7032E5F064A5702CC2AF758170F54B
:10FF1000B112FC5FE4F591F592F593F594F595F542
:10FF200096F5B17FD07E0712FD28E4F5B1750A0978
:10FF3000750B0612FDA702FDEE0000000000000098
:10FF400000000000000000000000000000000000B1
:10FF500000000000000000000000000000000000A1
:10FF60000000000000000000000000000000000091
:10FF70000000000000000000000000000000000081
:10FF80000000000000000000000000000000000071
:10FF90000000000000000000000000000000000061
:10FFA0000000000000000000000000000000000051
:10FFB0000000000000000000000000000000000041
:10FFC0000000000000000000000000000000000031
:10FFD0000000000000000000000000000000000021
:10FFE0000000000000000000000000000000000011
:10FFF0000000000000000000000000000000000001
:00000001FF

## idk if i did this right lol

Nono9k, 2024-04-27:
I used this tool lol, these cheaper kbs seem to be unrealible in terms of what you get lol

carlossless, 2024-04-27:
@Nono9k thanks for the dump. I can see from it that the bootloader is indeed the same as in #66 (`3e0ebd0c440af5236d7ff8872343f85d`), and the other md5 was for the full dump.

I'll close the issue as it seems no additional modifications are necessary.

Would you mind sharing a link to that temu link? $3 sounds like a steal!

closed 2024-04-27 by carlossless

Nono9k, 2024-04-27:
South Africa only link I think only sadly, it's a weird new user deal that they push a ton of ads for