karstx/sinowisp #83: [device-report] MAGEGEE STAR 61

leandrofriedrich, 2024-07-14:
Hi, i'm trying to get this software to work with my Star61 for purposes of dumping/analyzing the stock firmware and/or CFW, i tried all sorts of command combinations but could not get anything to work. Any chance i can get some help finding out the necessary infos?

## Device Info

- Sinowealth Device: _MAGEGEE STAR 61_
- IC Label: _example `BYK901`_
- Product Page: _https://www.magegee.com/products2.html?pid=565745&_t=1697698042_

## Part Info

```
firmware_size: unknown at the moment
vendor_id: 0x258a
product_id: 0x013b
bootloader_size: unknown at the moment # necessary if not default, otherwise remove this line
page_size: unknown at the momen t# necessary if not default, otherwise remove this line
isp_usage_page: unknown at the moment # necessary if not default, otherwise remove this line
isp_usage: unknown at the moment # necessary if not default, otherwise remove this line
isp_index: unknown at the moment # necessary if not default, otherwise remove this line
```

## Operations Tested

- [ ] Read
- [ ] Write

## Platforms Tested

- [ ] linux
- [ ] macos
- [x ] windows

## Checksums

- Stock Firmware MD5: `deadbeefdeadbeefdeadbeefdeadbeef`
- Bootloader MD5: `beefcafebeefcafebeefcafebeefcafe` _(shown when running `sinowealth-kb-tool read -b ...`)_

## HID Dump

A dump from [usbhid-dump](https://github.com/DIGImend/usbhid-dump), [win-hid-dump](https://github.com/todbot/win-hid-dump) or [mac-hid-dump](https://github.com/todbot/mac-hid-dump)

<details>
<summary>HID Tool Output</summary>

```
# NuPhy Air60 using win-hid-dump
...
WinHIDdump:
258A:013B: Sino Wealth - 61Keyboard
PATH:\\?\hid#vid_258a&pid_013b&mi_01&col07#9&1c962194&0&0006#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  06  00  FF  09  01  A1  01  85  0B  09  00  15  00  25  FF  35
  00  45  00  65  00  55  00  75  08  95  7E  B1  02  C1  00
  (31 bytes)
258A:013B: Sino Wealth - 61Keyboard
PATH:\\?\hid#vid_258a&pid_013b&mi_01&col06#9&1c962194&0&0005#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  06  00  FF  09  01  A1  01  85  0A  09  00  15  00  25  FF  35
  00  45  00  65  00  55  00  75  08  95  29  B1  02  C1  00
  (31 bytes)
258A:013B: Sino Wealth - 61Keyboard
PATH:\\?\hid#vid_258a&pid_013b&mi_01&col01#9&1c962194&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  05  01  09  80  A1  01  85  01  19  81  29  83  15  00  25  01
  35  00  45  01  65  00  55  00  75  01  95  03  81  02  95  05
  81  03  C1  00
  (36 bytes)
258A:013B: Sino Wealth - 61K
PATH:\\?\hid#vid_258a&pid_013b&mi_01&col02#9&1c962194&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  05  0C  09  01  A1  01  85  02  15  00  25  01  35  00  45  01
  65  00  55  00  75  01  95  10  81  03  C1  00
  (28 bytes)
258A:013B: Sino Wealth - 61Keyboard
PATH:\\?\hid#vid_258a&pid_013b&mi_01&col03#9&1c962194&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  06  00  FF  09  01  A1  01  85  05  15  00  25  01  35  00  45
  01  65  00  55  00  75  01  95  28  B1  03  C1  00
  (29 bytes)
258A:013B: Sino Wealth - 61Keyboard
PATH:\\?\hid#vid_258a&pid_013b&mi_00#9&2fc63dd2&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
DESCRIPTOR:
  05  01  09  06  A1  01  05  07  19  E0  29  E7  15  00  25  01
  35  00  45  01  65  00  55  00  75  01  95  08  81  02  95  38
  81  03  05  08  19  01  29  05  95  05  91  02  95  03  91  03
  C1  00
  (50 bytes)
...
258A:013B: Sino Wealth -
PATH:\\?\hid#vid_258a&pid_013b&mi_01&col09#9&1c962194&0&0008#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  06  00  FF  09  01  A1  01  85  0C  09  00  15  00  25  FF  35
  00  45  00  65  00  55  00  75  08  96  80  07  B1  02  C1  00

  (32 bytes)
258A:013B: Sino Wealth - 61Keyboard
PATH:\\?\hid#vid_258a&pid_013b&mi_01&col04#9&1c962194&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
DESCRIPTOR:
  05  01  09  06  A1  01  85  06  05  07  19  04  29  70  15  00
  25  01  35  00  45  01  65  00  55  00  75  01  95  6D  81  02
  95  0B  81  03  C1  00
  (38 bytes)
258A:013B: Sino Wealth - 61Keyboard
PATH:\\?\hid#vid_258a&pid_013b&mi_01&col08#9&1c962194&0&0007#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  05  01  09  02  A1  01  85  0D  09  01  A1  00  05  09  19  01
  29  05  15  00  25  01  35  00  45  01  65  00  55  00  75  01
  95  05  81  02  95  03  81  03  05  01  09  30  26  FF  7F  45
  00  75  10  95  01  81  06  09  31  81  06  09  38  25  7F  75
  08  81  06  05  0C  0A  38  02  81  06  C1  00  C1  00
  (78 bytes)
258A:013B: Sino Wealth - 61Keyboard
PATH:\\?\hid#vid_258a&pid_013b&mi_01&col05#9&1c962194&0&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  06  00  FF  09  01  A1  01  85  09  09  00  15  00  25  FF  35
  00  45  00  65  00  55  00  75  08  96  F8  01  B1  02  C1  00

  (32 bytes)

...
```

</details>

## PCB Photos

_If possible, include photos of your device PCB clearly showing MCU and wireless IC labels_

leandrofriedrich, 2024-07-14:
it looks like someone else tried looking into this too before (just seen it now so sry4duplicate) but couldnt and then dipped, so maybe we can try getting this to work again

carlossless, 2024-07-14:
@leandrofriedrich interesting, it's selecting the wrong TLC, the ISP report is on Col03 which should be the first one (and not require any additional options to the ones you already specified), but it selects Col05...

What does `sinowealth-kb-tool.exe list` give you?

leandrofriedrich, 2024-07-14:
@carlossless hi, this is the output of that command
```

INFO  [sinowealth_kb_tool::isp] Listing all connected HID devices...
INFO  [sinowealth_kb_tool::isp] \\?\HID#VID_046D&PID_C018#8&114ddd0f&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}: ID 046d:c018 manufacturer="Logitech" product="Logitech USB Optical Mouse" usage_page=0x0001 usage=0x0002
INFO  [sinowealth_kb_tool::isp] \\?\HID#VID_12D1&PID_3A07&MI_03#8&12fa93ad&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}: ID 12d1:3a07 manufacturer="bestechnic" product="HUAWEI USB-C HEADSET" usage_page=0x000c usage=0x0001
INFO  [sinowealth_kb_tool::isp] \\?\HID#VID_258A&PID_013B&MI_00#8&23ea0972&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\KBD: ID 258a:013b manufacturer="Sino Wealth" product="61Keyboard" usage_page=0x0001 usage=0x0006
INFO  [sinowealth_kb_tool::isp] \\?\HID#VID_258A&PID_013B&MI_01&Col01#8&12cbb0&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}: ID 258a:013b manufacturer="Sino Wealth" product="61Keyboard" usage_page=0x0001 usage=0x0080
INFO  [sinowealth_kb_tool::isp] \\?\HID#VID_258A&PID_013B&MI_01&Col02#8&12cbb0&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}: ID 258a:013b manufacturer="Sino Wealth" product="61Keyboard" usage_page=0x000c usage=0x0001
INFO  [sinowealth_kb_tool::isp] \\?\HID#VID_258A&PID_013B&MI_01&Col03#8&12cbb0&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}: ID 258a:013b manufacturer="Sino Wealth" product="61Keyboard" usage_page=0xff00 usage=0x0001
INFO  [sinowealth_kb_tool::isp] \\?\HID#VID_258A&PID_013B&MI_01&Col04#8&12cbb0&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}\KBD: ID 258a:013b manufacturer="Sino Wealth" product="61Keyboard" usage_page=0x0001 usage=0x0006
INFO  [sinowealth_kb_tool::isp] \\?\HID#VID_258A&PID_013B&MI_01&Col05#8&12cbb0&0&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}: ID 258a:013b manufacturer="Sino Wealth" product="61Keyboard" usage_page=0xff00 usage=0x0001
INFO  [sinowealth_kb_tool::isp] \\?\HID#VID_258A&PID_013B&MI_01&Col06#8&12cbb0&0&0005#{4d1e55b2-f16f-11cf-88cb-001111000030}: ID 258a:013b manufacturer="Sino Wealth" product="61Keyboard" usage_page=0xff00 usage=0x0001
INFO  [sinowealth_kb_tool::isp] \\?\HID#VID_258A&PID_013B&MI_01&Col07#8&12cbb0&0&0006#{4d1e55b2-f16f-11cf-88cb-001111000030}: ID 258a:013b manufacturer="Sino Wealth" product="61Keyboard" usage_page=0xff00 usage=0x0001
INFO  [sinowealth_kb_tool::isp] \\?\HID#VID_258A&PID_013B&MI_01&Col08#8&12cbb0&0&0007#{4d1e55b2-f16f-11cf-88cb-001111000030}: ID 258a:013b manufacturer="Sino Wealth" product="61Keyboard" usage_page=0x0001 usage=0x0002
INFO  [sinowealth_kb_tool::isp] \\?\HID#VID_258A&PID_013B&MI_01&Col09#8&12cbb0&0&0008#{4d1e55b2-f16f-11cf-88cb-001111000030}: ID 258a:013b manufacturer="Sino Wealth" product="61Keyboard" usage_page=0xff00 usage=0x0001
INFO  [sinowealth_kb_tool::isp] Found 12 devices

```

carlossless, 2024-07-15:
Oh, I just realized that the non-isp hid devices are unsorted - that's a bug. I'll be fixing it shortly.

In the meantime, try using:
```sh
DEBUG=1 ./sinowealth-kb-tool.exe read --vendor_id 0x258a --product_id 0x013b --isp_usage_page 0xff00 --isp_usage 0x0001 isp_index 1 --firmware_size 61440 dump.hex
# OR
DEBUG=1 ./sinowealth-kb-tool.exe read --vendor_id 0x258a --product_id 0x013b --isp_usage_page 0xff00 --isp_usage 0x0001 isp_index 2 --firmware_size 61440 dump.hex
# OR
DEBUG=1 ./sinowealth-kb-tool.exe read --vendor_id 0x258a --product_id 0x013b --isp_usage_page 0xff00 --isp_usage 0x0001 isp_index 3 --firmware_size 61440 dump.hex
# OR
DEBUG=1 ./sinowealth-kb-tool.exe read --vendor_id 0x258a --product_id 0x013b --isp_usage_page 0xff00 --isp_usage 0x0001 isp_index 4 --firmware_size 61440 dump.hex
```

One of them should work for you

leandrofriedrich, 2024-07-15:
@carlossless finally had a chance to test them, they all give me the same "0x00000001 Incorrect function" error :/

carlossless, 2024-07-16:
@leandrofriedrich can you also try with `--isp_index 0` then? (perhaps the enumeration order changed). I'm also curious what the debug logs return for each invocation, particularly `DEBUG [sinowealth_kb_tool::isp] Found Device: ...`

leandrofriedrich, 2024-07-16:
@carlossless hi, sorry for answering so late, here you go i put them all in a pastebin
https://pastebin.com/87CACKFM

carlossless, 2024-07-16:
@leandrofriedrich thanks! I see it returns a different error for the correct collection (Col3):
```
DEBUG [sinowealth_kb_tool::isp] Error: hidapi error: HidD_SetFeature: (0x0000001F) A device attached to the system is not functioning.
```

Unfortunately, this one indicates that there's something wrong with the implementation of this function within the firmware :/.

If there was some other means to get a firmware dump (possibly [gashtaan/sinowealth-8051-dumper](https://github.com/gashtaan/sinowealth-8051-dumper)), analyzing it could give us some clues on whether it's actually broken or it just expects a different payload.

cinnaooon, 2025-04-23:
Hello, i have the same Keyboard. Can i help or contribute in any way?

carlossless, 2025-04-24:
@cinnaooon are you running into the same issues as @leandrofriedrich?

cinnaooon, 2025-04-25:
I need to wait a few days until i got all equipment i need

cinnaooon, 2025-04-25:
Sorry, but what should i put into the path option when i execute it?

carlossless, 2025-04-25:
@cinnaooon set it to `--firmware_size 61440`

cinnaooon, 2025-04-25:
Reading works!

cinnaooon, 2025-04-25:
Uhm i just realized that my Keyboard has a different product id. Is that normal?

carlossless, 2025-04-25:
@cinnaooon it's not uncommon for these types of keyboards, but you certainly have different firmware than @leandrofriedrich, possibly a different hardware revision also.

Could you open up a [device report](https://github.com/carlossless/sinowealth-kb-tool/issues/new?template=device-report.md) with details of your device? (the template should guide you)

carlossless, 2025-04-25:
@cinnaooon I can see from your [report](https://github.com/carlossless/sinowealth-kb-tool/issues/101) that it's a different keyboard from the one mentioned here. `MK-STAR61 != STAR61` and it explains the difference in product_id (although not all manufacturers give different products different product_ids)

cinnaooon, 2025-04-26:
I don't know. I didn't find just a "STAR61" Keyboard anywhere online, just one with MK. I guess it's different because there are like four different color Versions as seen [here](https://www.magegee.com/products.html?pkeyword=star) on the sellers page.