karstx/sinowisp #95: [device-report] GIBKEY G68

mpaterakis, 2024-12-18:
## Device Info

- Sinowealth Device: `SH68F90A`
- IC Label: `BYK916`
- Product Page: https://www.aliexpress.com/item/1005007780444262.html

## Part Info

```
firmware_size: 61440
vendor_id: 0x258a
product_id: 0x0049
```

## Operations Tested

- [x] Read
- [ ] Write

## Platforms Tested

- [x] linux
- [ ] macos
- [x] windows

## Checksums

- Stock Firmware MD5: `deadbeefdeadbeefdeadbeefdeadbeef`
- Bootloader MD5: `beefcafebeefcafebeefcafebeefcafe` _(shown when running `sinowealth-kb-tool read -b ...`)_

## HID Dump

<details>
<summary>HID Tool Output</summary>

```
> usbhid-dump --model 258a:0049

001:020:002:DESCRIPTOR         1734553559.468228
 05 01 09 00 A1 01 15 00 26 FF 00 19 00 29 08 95
 40 75 08 81 02 19 01 29 08 95 40 75 08 91 02 C0

001:020:000:DESCRIPTOR         1734553559.471237
 05 01 09 06 A1 01 85 01 05 07 19 00 29 7C 15 00
 25 01 95 78 75 01 81 00 25 01 95 05 75 01 05 08
 19 01 29 05 91 02 95 01 75 03 91 03 C0 06 00 FF
 09 00 A1 01 85 02 19 00 2A FF 00 15 00 26 FF 00
 75 08 95 1F 91 00 19 00 2A FF 00 81 00 C0 05 01
 09 02 A1 01 85 03 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 95 05 75 01 81 02 95 01 75 03 81 03
 05 01 09 30 09 31 09 38 15 81 25 7F 75 08 95 03
 81 06 05 0C 0A 38 02 95 01 81 06 C0 C0 05 0C 09
 01 A1 01 85 04 19 01 2A A0 02 15 01 26 A0 02 95
 01 75 10 81 00 C0 05 01 09 80 A1 01 85 05 19 81
 29 83 25 01 75 01 95 03 81 02 95 05 81 01 C0 05
 01 09 0C A1 01 85 06 15 00 25 01 09 0C 95 01 75
 01 81 06 75 07 81 03 C0

001:020:001:DESCRIPTOR         1734553559.474214
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 95 01 75 08 81 01 95 05 75 01
 05 08 19 01 29 05 91 02 95 01 75 03 91 01 95 06
 75 08 15 00 26 FF 00 05 07 19 00 2A FF 00 81 00
 C0
```
</details>

I can't get a dump from this keyboard.

- When using Windows (2 different machines, multiple USB cables and ports) I am getting a very quick read process (less than 10 seconds) but I never get a "Found regular device" message. It always says nothing found. Listing works.
- When using Ubuntu LTS 24 I get an hid error. Listing works.
- When using a Steam Deck I get this, but very slowly (takes multiple minutes): 
```
> DEBUG=1 ./sinowealth-kb-tool read --vendor_id 0x258A --product_id 0x0049 --firmware_size 61440 nand.bin
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
DEBUG [sinowealth_kb_tool::isp] Found Device: "1-1.1:1.1"
DEBUG [sinowealth_kb_tool::isp] Opening: "1-1.1:1.1"
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Waiting for ISP device...
INFO  [sinowealth_kb_tool::isp] ISP device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
DEBUG [sinowealth_kb_tool::isp] Found Device: "1-1.1:1.1"
DEBUG [sinowealth_kb_tool::isp] Opening: "1-1.1:1.1"
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Waiting for ISP device...
INFO  [sinowealth_kb_tool::isp] ISP device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 3/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
DEBUG [sinowealth_kb_tool::isp] Found Device: "1-1.1:1.1"
DEBUG [sinowealth_kb_tool::isp] Opening: "1-1.1:1.1"
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Waiting for ISP device...
INFO  [sinowealth_kb_tool::isp] ISP device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 4/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
DEBUG [sinowealth_kb_tool::isp] Found Device: "1-1.1:1.1"
DEBUG [sinowealth_kb_tool::isp] Opening: "1-1.1:1.1"
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Waiting for ISP device...
INFO  [sinowealth_kb_tool::isp] ISP device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 5/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
DEBUG [sinowealth_kb_tool::isp] Found Device: "1-1.1:1.1"
DEBUG [sinowealth_kb_tool::isp] Opening: "1-1.1:1.1"
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Waiting for ISP device...
INFO  [sinowealth_kb_tool::isp] ISP device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 6/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
DEBUG [sinowealth_kb_tool::isp] Found Device: "1-1.1:1.1"
DEBUG [sinowealth_kb_tool::isp] Opening: "1-1.1:1.1"
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Waiting for ISP device...
INFO  [sinowealth_kb_tool::isp] ISP device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 7/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
DEBUG [sinowealth_kb_tool::isp] Found Device: "1-1.1:1.1"
DEBUG [sinowealth_kb_tool::isp] Opening: "1-1.1:1.1"
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Waiting for ISP device...
INFO  [sinowealth_kb_tool::isp] ISP device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
```

Hope you can help with this. :pray:

mpaterakis, 2024-12-20:
_Updated my original post with images and IC info._

**I can't get a dump from this keyboard.**

- When using Windows (2 different machines, multiple USB cables and ports) I am getting a very quick read process (less than 10 seconds) but I never get a "Found regular device" message. It always says nothing found. Listing works.
- When using Ubuntu LTS 24 I get an hid error. Listing works.
- When using a Steam Deck I get this, but very slowly (takes multiple minutes): 
```
> DEBUG=1 ./sinowealth-kb-tool read --vendor_id 0x258A --product_id 0x0049 --firmware_size 61440 nand.bin
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
DEBUG [sinowealth_kb_tool::isp] Found Device: "1-1.1:1.1"
DEBUG [sinowealth_kb_tool::isp] Opening: "1-1.1:1.1"
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Waiting for ISP device...
INFO  [sinowealth_kb_tool::isp] ISP device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
DEBUG [sinowealth_kb_tool::isp] Found Device: "1-1.1:1.1"
DEBUG [sinowealth_kb_tool::isp] Opening: "1-1.1:1.1"
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Waiting for ISP device...
INFO  [sinowealth_kb_tool::isp] ISP device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 3/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
DEBUG [sinowealth_kb_tool::isp] Found Device: "1-1.1:1.1"
DEBUG [sinowealth_kb_tool::isp] Opening: "1-1.1:1.1"
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Waiting for ISP device...
INFO  [sinowealth_kb_tool::isp] ISP device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 4/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
DEBUG [sinowealth_kb_tool::isp] Found Device: "1-1.1:1.1"
DEBUG [sinowealth_kb_tool::isp] Opening: "1-1.1:1.1"
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Waiting for ISP device...
INFO  [sinowealth_kb_tool::isp] ISP device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 5/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
DEBUG [sinowealth_kb_tool::isp] Found Device: "1-1.1:1.1"
DEBUG [sinowealth_kb_tool::isp] Opening: "1-1.1:1.1"
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Waiting for ISP device...
INFO  [sinowealth_kb_tool::isp] ISP device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 6/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
DEBUG [sinowealth_kb_tool::isp] Found Device: "1-1.1:1.1"
DEBUG [sinowealth_kb_tool::isp] Opening: "1-1.1:1.1"
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Waiting for ISP device...
INFO  [sinowealth_kb_tool::isp] ISP device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 7/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
DEBUG [sinowealth_kb_tool::isp] Found Device: "1-1.1:1.1"
DEBUG [sinowealth_kb_tool::isp] Opening: "1-1.1:1.1"
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Waiting for ISP device...
INFO  [sinowealth_kb_tool::isp] ISP device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
```

Here's the list output (keep in mind I have 2 keyboards connected):
```
INFO  [sinowealth_kb_tool::isp] Listing all connected HID devices...
INFO  [sinowealth_kb_tool::isp] 1-1.1:1.0: ID 258a:0049 manufacturer="ZXWMicroChip" product="ZXW-KEYBOARD"
INFO  [sinowealth_kb_tool::isp] 1-1.1:1.1: ID 258a:0049 manufacturer="ZXWMicroChip" product="ZXW-KEYBOARD"
INFO  [sinowealth_kb_tool::isp] 1-1.1:1.2: ID 258a:0049 manufacturer="ZXWMicroChip" product="ZXW-KEYBOARD"
INFO  [sinowealth_kb_tool::isp] 1-1.2:1.0: ID 1e7d:2dcd manufacturer="ROCCAT" product="ROCCAT Kone Pure Owl-Eye"
INFO  [sinowealth_kb_tool::isp] 1-1.2:1.1: ID 1e7d:2dcd manufacturer="ROCCAT" product="ROCCAT Kone Pure Owl-Eye"
INFO  [sinowealth_kb_tool::isp] 1-1.3:1.0: ID 25a7:fa70 manufacturer="Compx" product="2.4G Wireless Receiver"
INFO  [sinowealth_kb_tool::isp] 1-1.3:1.1: ID 25a7:fa70 manufacturer="Compx" product="2.4G Wireless Receiver"
INFO  [sinowealth_kb_tool::isp] 3-3:1.0: ID 28de:1205 manufacturer="Valve Software" product="Steam Deck Controller"
INFO  [sinowealth_kb_tool::isp] 3-3:1.1: ID 28de:1205 manufacturer="Valve Software" product="Steam Deck Controller"
INFO  [sinowealth_kb_tool::isp] 3-3:1.2: ID 28de:1205 manufacturer="Valve Software" product="Steam Deck Controller"
INFO  [sinowealth_kb_tool::isp] Found 10 devices
```

Hope you can help with this. :pray:

carlossless, 2024-12-20:
@mpaterakis there's no collection with `usage_page == 0xff00 && usage == 0x0001` in the HID report of your keyboard, but there is one with `usage == 0x0000` which could be the ISP mode trigger.

On one of your machines, try either:
```sh
sinowealth-kb-tool read --vendor_id 0x258A --product_id 0x0049 --firmware_size 61440 --isp_usage 0x00 --isp_iface_num 0 dump.hex
# or
sinowealth-kb-tool read --vendor_id 0x258A --product_id 0x0049 --firmware_size 61440 --isp_usage 0x00 dump.hex
```

>  I get an hid error

What's the error you got?

mpaterakis, 2024-12-20:
Thank you so much for the quick reply! I tried your suggestion on all my systems, and here are the results:

- `sinowealth-kb-tool read --vendor_id 0x258A --product_id 0x0049 --firmware_size 61440 --isp_usage 0x00 --isp_iface_num 0 dump.hex`: 
  - Windows: Same behavior as I was previously getting on the Steam Deck. "Found regular device." -> "Entering ISP mode..." -> "Regular device not found." -> "Trying ISP device..." and then it fails. This is progress at least, it never got to finding the regular device before.
  - SteamOS: No change besides the debug log showing "1-1.1:1.0" as Device
  - Ubuntu LTS 24: Same as SteamOS, besides the debug log showing "2-1.1:1.0" as Device

- `sinowealth-kb-tool read --vendor_id 0x258A --product_id 0x0049 --firmware_size 61440 --isp_usage 0x00 dump.hex`
  - Windows: Exactly the same as before, I'm getting "Regular device didn't come up..." and it keeps trying to find the ISP device with no success.
  - SteamOS: Exactly the same as the output on my previous comment.
  - Ubuntu LTS 24: Same as SteamOS, besides the debug log showing "2-1.1:1.1" as Device

_Note:_ Ubuntu 24 LTS was throwing `thread 'main' panicked at src/isp.rs:230:64: called Result::unwrap() on an Err value: HidApiError { message: "hid_error is not implemented yet" }`, but it went away by using sudo. Note that on SteamOS the behavior doesn't change when not using sudo, the program never throws any HID errors.

mpaterakis, 2024-12-22:
Not sure if it helps, but here's the `list` output on windows for this device, it's different than SteamOS:
```
INFO  [sinowealth_kb_tool::isp] \\?\HID#VID_258A&PID_0049&MI_00&Col01#7&1221a29f&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\KBD: ID 258a:0049 manufacturer="ZXWMicroChip" product="ZXWStdKeyboard" usage_page=0x0001 usage=0x0006
INFO  [sinowealth_kb_tool::isp] \\?\HID#VID_258A&PID_0049&MI_00&Col02#7&1221a29f&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}: ID 258a:0049 manufacturer="ZXWMicroChip" product="ZXWStdKeyboard" usage_page=0xff00 usage=0x0000
INFO  [sinowealth_kb_tool::isp] \\?\HID#VID_258A&PID_0049&MI_00&Col03#7&1221a29f&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}: ID 258a:0049 manufacturer="ZXWMicroChip" product="ZXWStdKeyboard" usage_page=0x0001 usage=0x0002
INFO  [sinowealth_kb_tool::isp] \\?\HID#VID_258A&PID_0049&MI_00&Col04#7&1221a29f&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}: ID 258a:0049 manufacturer="ZXWMicroChip" product="ZXWStdKeyboard" usage_page=0x000c usage=0x0001
INFO  [sinowealth_kb_tool::isp] \\?\HID#VID_258A&PID_0049&MI_00&Col05#7&1221a29f&0&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}: ID 258a:0049 manufacturer="ZXWMicroChip" product="ZXWStdKeyboard" usage_page=0x0001 usage=0x0080
INFO  [sinowealth_kb_tool::isp] \\?\HID#VID_258A&PID_0049&MI_00&Col06#7&1221a29f&0&0005#{4d1e55b2-f16f-11cf-88cb-001111000030}: ID 258a:0049 manufacturer="ZXWMicroChip" product="ZXWStdKeyboard" usage_page=0x0001 usage=0x000c
INFO  [sinowealth_kb_tool::isp] \\?\HID#VID_258A&PID_0049&MI_01#7&11b59b23&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\KBD: ID 258a:0049 manufacturer="ZXWMicroChip" product="ZXWConflictfree" usage_page=0x0001 usage=0x0006
INFO  [sinowealth_kb_tool::isp] \\?\HID#VID_258A&PID_0049&MI_02#7&358cd8e5&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}: ID 258a:0049 manufacturer="ZXWMicroChip" product="ZXWCustom" usage_page=0x0001 usage=0x0000
```

carlossless, 2024-12-22:
I now noticed the updated IC label which I haven't seen before, which makes me think that the mcu is unlikely a sinowealth part and less so that includes a ISP bootloader that could be interacted with by this tool.

Nevertheless, I made some changes in https://github.com/carlossless/sinowealth-kb-tool/tree/device/gibkey-g68 that you can build and try testing with:
```
DEBUG=1 sinowealth-kb-tool read --part gibkey-g68 dump.hex
```

mpaterakis, 2024-12-22:
Thank you for the changes, I will be trying them tomorrow!

I assumed the PID & VID matching the SH68xxxx chips would make it a sinowealth part too, didn't know it could match those by just having an ISP bootloader.

<details>
<summary>BIG sidenote inside, strap in:</summary>

Googling the IC code results in only 1 review of another keyboard, entirely in Russian, where the reviewer is just as confused about this chip as I am.

So I decided to email the address I found on the manual and ask about the part, any firmware files etc but never got a reply. I asked the seller the same thing and they told me the manufacturer refused to send them anything besides the Windows drivers (which is just a utility for remappping the keys and set lighting effects). I asked for the name of the actual manufacturer but haven't gotten a reply. I expect them to be just a random chinese factory though.

Finally, I tried using the Arduino dumper, but since the board doesn't have a clear JTAG header, (as you can see in the images) I tried tracing the chip pins to the ones that are at least exposed on it. I couldn't get a match to those, and I also found some weird ground pins on the chip (the right-most bottom pin, in particular) that I couldn't match to any of the sinowealth 8051 schematics out there. I don't know if it's just my lack of experience but it has been a bizzare journey, and your assumption holds quite some weight here.
</details>

carlossless, 2024-12-22:
Well yeah, the fact that it's using sinowealths registered VID could suggest that it's actually a sinowealth part, but the HID descriptor still lacks the ISP report (at least it does not appears how it does on all other devices encountered so far).

> the Windows drivers (which is just a utility for remappping the keys and set lighting effects)

This is what I am guessing the HID report id 0x02 is for, making it less likely that it could trigger ISP mode :(

> Finally, I tried using the Arduino dumper, but since the board doesn't have a clear JTAG header, (as you can see in the images) I tried tracing the chip pins to the ones that are at least exposed on [...]

If it's a sinowealth part, I would guess CON3 is JTAG. Are you sure they don't connect to any pins on the chip? Perhaps there are some passives in between?

I've seen many sinowealth keyboard (and mice) PCBs use the same ordering of pins, so my guess, if CON3 is JTAG, the pins could be (from left to right, as seen on the picture you shared):
1. GND
2. TDO
3. TMS
4. TDI
5. TCK
6. VDD

mpaterakis, 2024-12-23:
I used the artifacts from the github actions. So here are the results of the test:

SteamOS and Ubuntu: No difference compared to before. "Regular device found" then doesn't find the ISP and keeps retrying.
Windows:
```
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
DEBUG [sinowealth_kb_tool::isp] Found Device: "\\\\?\\HID#VID_258A&PID_0049&MI_00&Col02#7&1221a29f&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}" 0xff00 0x0000
DEBUG [sinowealth_kb_tool::isp] Opening: "\\\\?\\HID#VID_258A&PID_0049&MI_00&Col02#7&1221a29f&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}"
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
DEBUG [sinowealth_kb_tool::isp] Error: hidapi error: HidD_SetFeature: (0x00000001) Incorrect function.
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
```
...and so on. 
I never ran DEBUG=1 on Windows before, turns out all versions (0.1.1, master, gibkey-g68) throw this hid error. But _why only on Windows?_

> If it's a sinowealth part, I would guess CON3 is JTAG. Are you sure they don't connect to any pins on the chip? Perhaps there are some passives in between?

All of the CON3 pins (except for one, which I confirmed to be VDD) connect to chip pins, that's how I found the weird ground pins I mentioned before too. But I couldn't figure out what was what, the pin locations I was getting didn't match any pinout I found online.
I will try your recommendation, thank you again.

mpaterakis, 2024-12-23:
Tested the arduino dumper and nothing. I even tried changing the pins around but nothing. I'll stop experimenting with that so I don't destroy my board though. 😅

mpaterakis, 2024-12-24:
I looked into the drivers for this thing and noticed that it always tries to use "MI_02". To be completely honest I have no idea what that is, I'm not well versed in USB stuff, but the program calls `VID_%s&PID_%s&MI_02`, for all types of keyboards that use this type of program. I thought I should mention it in case it's related to the `isp_report_id` you added in your latest commit.

carlossless, 2024-12-25:
According to https://learn.microsoft.com/en-us/windows-hardware/drivers/hid/hidclass-hardware-ids-for-top-level-collections MI_02 should refer to the interface number, so it's targeting a different report than where Report ID = 2 resides (it's in interface 1).

For both of our reference, the HID descriptor for the 02 interface is:
```
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x00,        // Usage (Undefined)
0xA1, 0x01,        // Collection (Application)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x19, 0x00,        //   Usage Minimum (Undefined)
0x29, 0x08,        //   Usage Maximum (Multi-axis Controller)
0x95, 0x40,        //   Report Count (64)
0x75, 0x08,        //   Report Size (8)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x19, 0x01,        //   Usage Minimum (Pointer)
0x29, 0x08,        //   Usage Maximum (Multi-axis Controller)
0x95, 0x40,        //   Report Count (64)
0x75, 0x08,        //   Report Size (8)
0x91, 0x02,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              // End Collection
```

mpaterakis, 2025-01-05:
First of all, happy new year!

In case someone else stumbles across this thread, I made a fully functional, open source alternative to the official configuration software for this keyboard. The official software is riddled with bugs and GIBKEY support refused to give me any source code or schematics.

So I did it myself: https://github.com/mpaterakis/GIBKEY-G68-Config

I understand that this not really related to the dumper, but at least it's a solution to my own problem and wanted to mention it.

carlossless, 2025-01-11:
@mpaterakis I'm going to close this issue since this keyboard (or at least its firmware) does not seem to be compatible with `sinowealth-kb-tool`. Feel free to continue adding any new findings you come across here though!

closed 2025-01-11 by carlossless