karstx/sinowisp #96: [device-report] Aula F75

cyrkul, 2024-12-26:
## Device Info

- Sinowealth Device:  `Dont know how to check`
- IC Label:  `Didn't open it up yet`
- Product Page: [https://aulagear.com/products/aula-f75](https://aulagear.com/products/aula-f75)

## Part Info

```
firmware_size: 61440
vendor_id: 0x258a
product_id: 0x010c
```

## Operations Tested

- [x] Read
- [ ] Write

## Platforms Tested

- [x] linux
- [ ] macos
- [ ] windows

## Checksums

- Stock Firmware MD5: `916e775e5cd88373fda373df25859f2e`
- Bootloader MD5: `3e0ebd0c440af5236d7ff8872343f85d`
- Full MD5: `9be3be27a1ff0a7ff806518f0e1d3aa9`

## HID Dump

A dump from [usbhid-dump](https://github.com/DIGImend/usbhid-dump), [win-hid-dump](https://github.com/todbot/win-hid-dump) or [mac-hid-dump](https://github.com/todbot/mac-hid-dump)

<details>
<summary>HID Tool Output</summary>

```
# NuPhy Air60 using win-hid-dump
...
usbhid-dump --model=258a:010c
001:034:001:DESCRIPTOR         1735242590.233973
 06 01 00 09 80 A1 01 85 01 19 81 29 83 15 00 25
 01 95 03 75 01 81 02 95 01 75 05 81 01 C0 05 0C
 09 01 A1 01 85 02 19 00 2A FF 02 15 00 26 FF 7F
 95 01 75 10 81 00 C0 06 00 FF 09 01 A1 01 85 03
 15 00 26 FF 00 09 2F 75 08 95 03 81 02 C0 05 01
 09 06 A1 01 85 04 05 07 19 04 29 70 15 00 25 01
 75 01 95 78 81 02 C0 06 00 FF 09 01 A1 01 85 05
 15 00 26 FF 00 19 01 29 02 75 08 95 05 B1 02 C0
 06 00 FF 09 01 A1 01 85 06 15 00 26 FF 00 75 08
 95 07 09 02 81 00 15 00 26 FF 00 19 01 29 02 75
 08 96 07 02 B1 02 C0 05 01 09 02 A1 01 85 07 09
 01 A1 00 05 09 15 00 25 01 19 01 29 05 75 01 95
 05 81 02 95 03 81 01 05 01 16 00 80 26 FF 7F 09
 30 09 31 75 10 95 02 81 06 15 81 25 7F 09 38 75
 08 95 01 81 06 05 0C 0A 38 02 95 01 81 06 C0 C0

001:034:000:DESCRIPTOR         1735242590.237333
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 95 08 75 01 81 02 95 01 75 08 81 03 95 06 75 08
 15 00 26 FF 00 05 07 19 00 2A FF 00 81 00 25 01
 95 05 75 01 05 08 19 01 29 05 91 02 95 01 75 03
 91 03 C0
...
```

</details>

## PCB Photos
Not opened keyboard yet

carlossless, 2025-01-11:
@cyrkul thanks for the report. Will you be willing to open it up to check the IC labels?

cyrkul, 2025-01-15:
Hello @carlossless,

I finally found time to open it and take a picture - I updated the post.
I still don't know how to check Sinowealth Device and if all information I gathered is correct.

carlossless, 2025-01-15:
@cyrkul thanks, everything looks good. Interesting to see there's another package variant for the BK3632.

closed 2025-01-15 by carlossless