karstx/sinowisp #100: [device-report] Redragon Horus TKL K621

WilliamFernsV3, 2025-02-03:
## Device Info

- Sinowealth Device:`SH68F90A`
- IC Label: _example `BYK916
- Product Page: https://www.redragonzone.com/products/redragon-k621-horus

## Part Info

```
firmware_size: ??
vendor_id: 0x258a
product_id: 0x0049
bootloader_size: ??
page_size: ?
```

## Operations Tested

- [x] Read
- [ ] Write

## Platforms Tested

- [ ] linux
- [x] macos
- [ ] windows

## Checksums

- Stock Firmware MD5: `deadbeefdeadbeefdeadbeefdeadbeef`
- Bootloader MD5: `beefcafebeefcafebeefcafebeefcafe` _(shown when running `sinowealth-kb-tool read -b ...`)_

## HID Dump

A dump from [usbhid-dump](https://github.com/DIGImend/usbhid-dump), [win-hid-dump](https://github.com/todbot/win-hid-dump) or [mac-hid-dump](https://github.com/todbot/mac-hid-dump)

<details>
<summary>HID Tool Output</summary>

```
# Redragon Horus TKL K621 using mac-hid-dump

(for some reason it shows two, don't know why.)
...

258A 0049: BY Tech - Gaming Keyboard
DESCRIPTOR:
  06  01  00  09  80  a1  01  85  01  19  81  29  83  15  00  25
  01  95  03  75  01  81  02  95  01  75  05  81  01  c0  05  0c
  09  01  a1  01  85  02  19  00  2a  ff  02  15  00  26  ff  7f
  95  01  75  10  81  00  c0  06  00  ff  09  01  a1  01  85  03
  15  00  26  ff  00  09  2f  75  08  95  03  81  02  c0  05  01
  09  06  a1  01  85  04  05  07  19  04  29  70  15  00  25  01
  75  01  95  78  81  02  c0  06  00  ff  09  01  a1  01  85  05
  15  00  26  ff  00  19  01  29  02  75  08  95  05  b1  02  c0
  06  00  ff  09  01  a1  01  85  06  15  00  26  ff  00  19  01
  29  02  75  08  96  07  04  b1  02  c0  05  01  09  02  a1  01
  85  07  09  01  a1  00  05  09  15  00  25  01  19  01  29  05
  75  01  95  05  81  02  95  03  81  01  05  01  16  00  80  26
  ff  7f  09  30  09  31  75  10  95  02  81  06  15  81  25  7f
  09  38  75  08  95  01  81  06  05  0c  0a  38  02  95  01  81
  06  c0  c0
  (227 bytes)
258A 0049: BY Tech - Gaming Keyboard
DESCRIPTOR:
  05  01  09  06  a1  01  05  07  19  e0  29  e7  15  00  25  01
  95  08  75  01  81  02  95  01  75  08  81  03  95  06  75  08
  15  00  26  ff  00  05  07  19  00  2a  ff  00  81  00  25  01
  95  05  75  01  05  08  19  01  29  05  91  02  95  01  75  03
  91  03  c0
  (67 bytes)
...
```

</details>

## PCB Photos

(used ChatGPT OCR capabilities for getting identifiers)

Main microcontroller: 
BYK916 -00997A NSMYM 2224
[image]

Bluetooth chip: BK3632 BU305WXB
[image]

---

Problem: when reading, it finds the device, but then it keyboard powers off. Here is the output:

```
cargo run read \
    --vendor_id 0x258a \
    --product_id 0x0049 \
    --firmware_size 61440 \
    --bootloader_size 4096 \
    --page_size 2048 \
    --isp_iface_num 1 \
    --isp_usage_page 0x0001 \
    --isp_usage 0x0080 \
    --isp_index 1 \
    --reboot false \
    foobar.hex
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.07s
     Running `target/debug/sinowealth-kb-tool read --vendor_id 0x258a --product_id 0x0049 --firmware_size 61440 --bootloader_size 4096 ' '`
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
DEBUG [sinowealth_kb_tool::isp] Found Device: "DevSrvsID:4294975728" 0xff00 0x0001
DEBUG [sinowealth_kb_tool::isp] Opening: "DevSrvsID:4294975728"
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
DEBUG [sinowealth_kb_tool::isp] Error: hidapi error: IOHIDDeviceSetReport failed: (0xE00002ED) (iokit/common) device not responding
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 3/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 4/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Regular device didn't come up...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...

```

Any reason why this is happening? Here is the output when I list all devices:

```

INFO  [sinowealth_kb_tool::isp] Listing all connected HID devices...
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294969420: ID 0000:0000 manufacturer="Apple" product="Headset" usage_page=0x000c usage=0x0001
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294969531: ID 0000:0000 manufacturer="APPL" product="BTM" usage_page=0xff00 usage=0x0048
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294969871: ID 05ac:8104 manufacturer="Apple" product="" usage_page=0x0020 usage=0x008a
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294969872: ID 0000:0000 manufacturer="Apple" product="" usage_page=0xff0c usage=0x0005
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294969873: ID 0000:0000 manufacturer="Apple" product="" usage_page=0xff00 usage=0x0004
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294969877: ID 05ac:8104 manufacturer="Apple" product="" usage_page=0xff00 usage=0x0003
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294969880: ID 05ac:8104 manufacturer="Apple" product="" usage_page=0xff00 usage=0x0009
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294969881: ID 0000:0000 manufacturer="Apple" product="" usage_page=0xff0c usage=0x0001
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294969882: ID 05ac:8104 manufacturer="Apple" product="" usage_page=0xff00 usage=0x0005
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294969884: ID 0000:0000 manufacturer="Apple" product="" usage_page=0xff00 usage=0x00ff
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294969986: ID 0000:0000 manufacturer="Apple" product="Apple Internal Keyboard / Trackpad" usage_page=0x0001 usage=0x0006
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294969986: ID 0000:0000 manufacturer="Apple" product="Apple Internal Keyboard / Trackpad" usage_page=0x000c usage=0x0001
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294969986: ID 0000:0000 manufacturer="Apple" product="Apple Internal Keyboard / Trackpad" usage_page=0xff00 usage=0x0006
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294969988: ID 0000:0000 manufacturer="Apple" product="Apple Internal Keyboard / Trackpad" usage_page=0xff00 usage=0x0003
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294969990: ID 0000:0000 manufacturer="Apple" product="Apple Internal Keyboard / Trackpad" usage_page=0xff00 usage=0x000d
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294970000: ID 0000:0000 manufacturer="Apple" product="Apple Internal Keyboard / Trackpad" usage_page=0xff00 usage=0x005f
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294970005: ID 0000:0000 manufacturer="Apple" product="Apple Internal Keyboard / Trackpad" usage_page=0xff00 usage=0x000b
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294970057: ID 0000:0000 manufacturer="Apple" product="Apple Internal Keyboard / Trackpad" usage_page=0x0001 usage=0x0002
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294970057: ID 0000:0000 manufacturer="Apple" product="Apple Internal Keyboard / Trackpad" usage_page=0x0001 usage=0x0001
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294970057: ID 0000:0000 manufacturer="Apple" product="Apple Internal Keyboard / Trackpad" usage_page=0x000d usage=0x0005
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294970057: ID 0000:0000 manufacturer="Apple" product="Apple Internal Keyboard / Trackpad" usage_page=0xff00 usage=0x000c
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294970659: ID 05ac:0000 manufacturer="" product="Keyboard Backlight" usage_page=0xff00 usage=0x000f
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294975866: ID 258a:0049 manufacturer="BY Tech" product="Gaming Keyboard" usage_page=0x0001 usage=0x0080
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294975866: ID 258a:0049 manufacturer="BY Tech" product="Gaming Keyboard" usage_page=0x000c usage=0x0001
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294975866: ID 258a:0049 manufacturer="BY Tech" product="Gaming Keyboard" usage_page=0xff00 usage=0x0001
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294975866: ID 258a:0049 manufacturer="BY Tech" product="Gaming Keyboard" usage_page=0x0001 usage=0x0006
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294975866: ID 258a:0049 manufacturer="BY Tech" product="Gaming Keyboard" usage_page=0x0001 usage=0x0002
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294975866: ID 258a:0049 manufacturer="BY Tech" product="Gaming Keyboard" usage_page=0x0001 usage=0x0001
INFO  [sinowealth_kb_tool::isp] DevSrvsID:4294975867: ID 258a:0049 manufacturer="BY Tech" product="Gaming Keyboard" usage_page=0x0001 usage=0x0006
INFO  [sinowealth_kb_tool::isp] Found 29 devices
```

WilliamFernsV3, 2025-02-03:
This might be relevant. Maybe it is not configuring device beforehand on mac os. Some drivers (like those for mac os) might not configure on initialization. 

https://github.com/libusb/libusb/issues/45#issuecomment-67205507

carlossless, 2025-02-05:
@WilliamFernsV3 can you run `list` right after the ISP command is sent and "keyboard powers off" and share its output, without replugging the keyboard? Perhaps you can also run `lsusb` or `ioreg -p IOUSB -l -w 0` at the same point too and share its output.

apdarks, 2025-03-02:
Same issue with K642. 
```
"D:\sinowealth-kb-tool.exe" read --vendor_id 0x258A --product_id 0x0049 --firmware_size 61440 backup
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 2/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 3/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 4/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 5/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 6/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 7/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 8/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 9/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
INFO  [sinowealth_kb_tool::isp] Retrying... Attempt 10/10
INFO  [sinowealth_kb_tool::isp] Looking for vId:0x258a pId:0x0049
INFO  [sinowealth_kb_tool::isp] Found regular device. Entering ISP mode...
INFO  [sinowealth_kb_tool::isp] Regular device not found. Trying ISP device...
ERROR [sinowealth_kb_tool] Device not found
```

apdarks, 2025-04-05:
It worked on linux, I did it with K642 by doing:
sudo ./tool -r --vendor_id 0x258a --product_id 0x0049 --firmware_size 61440 full.hex.

carlossless, 2025-07-06:
I'm closing this due to inactivity. If anyone comes back to this issue, try using the latest version of this tool to complete the device report(s) and/or share more information about what the `list` command reports about your device.

closed 2025-07-06 by carlossless