karstx/sinowisp #113: [bug] Trying to dump as sh68f881 fails

baruch, 2025-08-02:
The command line says arguments for --platform are either sh68f881 or sh68f90 but the code inside checks for sh68f91. I don't know which one is correct or not, but when I changed the code as below I got the sh68f881 working.

The command that failed for me was:
```
sudo target/debug/sinowealth-kb-tool read --vendor_id 0x258a --product_id 0x013b --platform sh68f881  --firmware_size 61440 dump_f881.hex 

thread 'main' panicked at src/main.rs:435:18:
Invalid platform
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

The diff I created:

```
diff --git src/main.rs src/main.rs
index 7feb593..fb992e1 100644
--- src/main.rs
+++ src/main.rs
@@ -431,7 +431,7 @@ fn get_device_spec_from_matches(sub_matches: &ArgMatches) -> DeviceSpec {
     if let Some(platform_name) = platform_name {
         device_spec = match platform_name {
             "sh68f90" => Some(DEVICE_BASE_SH68F90),
-            "sh68f91" => Some(DEVICE_BASE_SH68F881),
+            "sh68f881" => Some(DEVICE_BASE_SH68F881),
             _ => panic!("Invalid platform"),
         }
     }
```

carlossless, 2025-08-02:
Thanks for the report! Goes to show how often I test this platform...

baruch, 2025-08-02:
My device can be read with both platforms, but I get a lot of zeroes at the end, so I assume this is closer to 881 than 90. You can probably test both platforms on the same device and just see that you get less data.